raspberrypi/documentation #1423: Bridged Access Point instructions do not work for me

Taikuh, 2020-03-01:
I understand this repo is for documentation, so I will try to keep this issue within that scope. I will further research and ask around the forums to find out more about the technical side of things.

The steps the current master's steps ([here](https://github.com/raspberrypi/documentation/blob/master/configuration/wireless/access-point.md#using-the-raspberry-pi-as-an-access-point-to-share-an-internet-connection-bridge)) nor the steps in the next major revision ([here](https://github.com/raspberrypi/documentation/blob/aaf97e8c5c85e1aeb33bdd5e4119e6330a2e2f51/configuration/wireless/access-point-bridged.md)) work for me. Using #1333 as reference, the following modifications have worked for me:

##Added

```
sudo apt install bridge-utils

sudo nano /etc/systemd/network/bridge-br0.network
```

File contents:

```
[Match]
Name=br0

[Network]
Address=192.168.1.3/24
Gateway=192.168.1.1
DNS=8.8.8.8
```

where:

* `Address` serves as the bridge `br0`'s static IP within my LAN. It is outside my router's DHCP range.

* `Gateway` is my router's IP address

The following addition is sourced from [here](https://github.com/SurferTim/documentation/blob/6bc583965254fa292a470990c40b145f553f6b34/configuration/wireless/access-point.md#using-the-raspberry-pi-as-an-access-point-to-share-an-internet-connection):

```
sudo nano /etc/network/interfaces
```

File contents:

```
auto br0
iface br0 inet manual
bridge_ports eth0 wlan0
```

##Modifications

The following two files ([here](https://github.com/SurferTim/documentation/blob/6bc583965254fa292a470990c40b145f553f6b34/configuration/wireless/access-point.md#using-the-raspberry-pi-as-an-access-point-to-share-an-internet-connection)) are not necessary for me. They don't seem to do anything and removing them doesn't seem to affect my end result.

```
/etc/systemd/network/bridge-br0.netdev
/etc/systemd/network/br0-member-eth0.network
```

Again, something is probably technically wrong from my end. It could be something to do with my router's ipv6 settings that I can't seem to disable. Hopefully this issue will serve as a record and hint for others facing similar problems. If there's any way for me to format or present this issue more clearly, please let me know.

lurch, 2020-03-02:
ping @epoch1970

epoch1970, 2020-03-02:
Using ifupdown is the step backwards we don't need at all. ifupdown makes installing bridge-utils necessary but it is obsolete (too).
Defining the bridge in systemd and in interfaces is redundant. 
Defining a static IP for the bridge is not generic.  Defining it in systemd is at odds with Raspbian where everything IP related is defined in dhcpcd.conf (and also with ifupdown since the static IP could be set in the venerable interfaces file as well...)
Pong.

Taikuh, 2020-03-02:
Thanks for the reply. I've been tracking other networking threads on the official forums where you've replied but my account isn't allowed to post yet. I hope it's fine to try to figure it out here.

I understand that I should be using systemd-networkd to only create the bridge and to use dhcpcd to manage its IP. I also understand I shouldn't be using /etc/network/interfaces. 

Following #1333, **br0** isn't created. It doesn't show up in ifconfig.  So I need to use bridge-utils and /etc/network/interfaces to build the bridge.

Now I have a somewhat functioning bridge from bridge-utils and network/interfaces. But it is not being assigned an IP, even though #1333 claims dhcpcd will automatically assign IP.  So I use systemd-networkd to manually assign IP.

The result is my first post above.  Now question, is there anything I'm overlooking even though I followed #1333 100% on a fresh install?  Which file's contents would you like to see to help debug?

epoch1970, 2020-03-02:
Go to the forums for support. I suggest to close this ticket and see if someone else fails to get #1333 to work.

Taikuh, 2020-03-03:
I found a [thread in the forums](https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=260711) that mirrored my situation exactly and used the solution provided in that thread.  I decided to [reply in that thread](https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=260711&p=1621092#p1621092) instead of creating a new one because of how similar the situations were.  We could continue our discussion there if you'd like.  Thanks!

(I suggest adding that thread's solution to a Troubleshooting section to #1333)

epoch1970, 2020-03-07:
So. I ran the howto again and I can confirm it is broken on Pi3b with the current release of Raspbian Lite.
I think the issue lies somewhere in the bowels of systemd or udev. Following the howto, the bridge is created in down state and stays that way. Adding a .network file matching br0 gets it up. However dhcpcd seems incapacitated. 
I am opposed to using systemd to manage the IP config of the bridge, and for now I have no solution to offer.
Anyone?

epoch1970, 2020-03-07:
Actually, I just found a solution. It is kludgy, but it works for me, and it is not related to systemd.
Follow the howto, but in dhcpcd.conf, have:
```
denyinterfaces eth0 wlan0
allowinterfaces br0
```
at the beginning of the file. Adding the allowinterfaces option somehow helps dhcpcd take over and it fetches an address for br0. Why oh why?

Less incorrect would be to have `denyinterfaces eth0 wlan0` at the beginning of the file, and `allowinterfaces br0` at the end of the file, so that global options defined in the conf file would apply to the br0 lease.

This change of behavior makes no sense to me. So I'm not sure I am ready to recommend that "solution"

Taikuh, 2020-03-07:
I can confirm this works on the latest Raspbian Lite on a Pi4. No need anymore for the extraneous `/etc/systemd/network/bridge-br0.network` file I mentioned above either.

This indeed is quite a strange 'solution'. For clarification, `interface br0` should be above all the default config lines? Something like:

```bash
denyinterfaces wlan0 eth0
interface br0

hostname
clientid
persistent
...
```

epoch1970, 2020-03-07:
Not really. AFAIK, dhcpcd treats stand-alone options, like 'hostname' here, as global. Then when it sees a stanza that opens a block, like 'interface' or 'ssid', any option defined under it is local to that block. Global options still apply. A block is closed by a blank line.

So 'interface br0' at the top of the line relies only on dhcpcd's built-in defaults and the lease would not reflect options added by the user. Not good.

It is just a bit awkward to say "add deyninterface to the beginning of the file" and "add interface br0 at the end of the file"... That's why I bunched the 2 together, but that is not correct.

epoch1970, 2020-03-07:
BTW, thanks for the heads-up on the howto. You were right all the way.

Taikuh, 2020-03-07:
> Less incorrect would be to have `denyinterfaces eth0 wlan0` at the beginning of the file,

Ah yes, thanks for the clarification. I just read too fast your previous comment and was thinking More incorrect. 

I think to speed things along with #1333, it would be best to just use the `interface br0` hack-solution, with an addendum that it is not best practice, but is the best practical method. This way, the official docs can sooner reflect something that works and is more up-to-date.  I think that's important because most other tutorials I found online about using the RPi as an AP in bridged or routed mode are also outdated and plain don't work.

epoch1970, 2020-03-07:
Yes I would align with this solution too. dhcpcd.conf is more "userland" than /etc/systemd/network, methinks.

Please define "soon". When I contributed the how-to, months ago, it was working. It decayed while in copy-proofing ;)

Taikuh, 2020-03-07:
Yes, I understand the copy-proofing section is a bit short-staffed. I didn't mean any rush nor to sound entitled.  I was getting frustrated when I opened the I issue a week ago but I found all the help I needed here, especially from you. I may have to ping you again in the future for raspi networking help ;) I'll leave this issue open for now for others to provide alternatives.

jobtijhuis, 2020-03-25:
I have a comment after trying to set up an access point in bridge mode (using raspberry pi 3B+ on Buster).
I think using systemd-networkd to set up the bridge and attach eth0 to the bridge is the way to go, but the instructions are not complete. This is because systemd-networkd is not enabled by default and thus will not link the ethernet port to the bridge if it isn't even started up (leaving eth0 in the DOWN state).
So there should also be something about systemctl start/enable systemd-networkd in the instructions (for me that fixed the raspberry not being connected to the internet).

jobtijhuis, 2020-03-26:
BTW it is also unclear what from the first (with DHCP server) part also applies to the second (bridging) part, for example the 'routing and masquerade'.

epoch1970, 2020-03-26:
In both of your comments, are you referring to the document at #1333?
It doesn't look so.

jobtijhuis, 2020-03-27:
I did indeed not look at closely at the contents of #1333 so I missed that the part about systemctl enable systemd-networkd was already in there. But I still think that the part about ipv4 forwarding is still needed to make the bridged version work, at least that was the case for me. I did not try #1333 from scratch again so I am not 100% sure.

JamesH65, 2020-04-24:
Should be fixed now.

closed 2020-04-24 by JamesH65