raspberrypi/documentation #1424: Change of behavior for device_tree_address?

andreiw, 2020-03-02:
If a DTB is to be loaded at the address specified by device_tree_address, when is this done relative to armstub? After armstub is loaded or before? Probably should be documented.

Pi 3 (and subsequently Pi 4) used to rely on the DTB appearing in memory after armstub being loaded. armstub would be the UEFI image, with an area left intentionally empty (that device_tree_address would be used to fill).

However, this no longer appears to work. The area is always empty, as if the dtb was placed before the armstub.

andreiw, 2020-03-02:
Dumping the armstub area shows start.elf recognised the desired address and placed dtb there:

  000000F0: 00 00 00 00 00 00 00 00-00 00 02 00 00 00 00 00  *................*
  00000100: 44 11 00 00 00 00 00 00-00 00 00 00 00 00 00 00  *D...............*

...corresponds to 0x20000. Addresses directly armstub seem fine.

andreiw, 2020-03-02:
I'm assuming the answer is going to be something along the lines of "don't do that anymore", so... can I rely on the DTB being loaded (by default) in a way that won't overlap armstub?

i.e. the following being guaranteed to be apart? I remember back in Pi 3 days the DTB being loaded at 0x100 by default... or some low overlapping address like 1MB?

armstub=RPI_EFI.fd
device_tree=bcm2711-rpi-4-b.dtb

lurch, 2020-03-02:
ping @pelwell

pelwell, 2020-03-02:
The load order is currently:
1. Load any initramfs high to the specified address. If no address, load to 0 and relocate because we don't know the size.
2. Load DTB where requested, or load to 0x100 then copy high
3. Load or copy the stub to 0.
4. Extract the preferred kernel and DTB address, moving the DTB if necessary.
5. Load the kernel.

When I say "currently", I'm happy with the setup now (it fixes a number of limitations of the previous flow), but I reserve the right to change it in the future.

Some possible solutions to your problem are:
a. Populate the DT address field in the stub and let the loader copy it there.
b. Just let the loader load it high and use the value in the DT address field.

andreiw, 2020-03-02:
Thanks. Could you confirm that this behavior is consistent across Pi 3 and Pi 4?

pelwell, 2020-03-02:
Yes it is - that bit of code is common to all Pis.

closed 2020-03-02 by andreiw