raspberrypi/documentation #1427: A SSH / UART dongle client for Android suggestion.

green-green-avk, 2020-03-04:
Um... Not so humble from my side. It's rather a suggestion.
I tried different SSH and UART dongle clients for Android... and found a lack of good input / rendering features.
But what do you think about this one:
https://play.google.com/store/apps/details?id=green_green_avk.anotherterm

Thank you for your patience in advance.

lurch, 2020-03-05:
Are you suggesting that this should be added to the list of clients on [this page](https://github.com/raspberrypi/documentation/blob/master/remote-access/ssh/android.md) ?

lurch, 2020-03-05:
Although @JamesH65 do you think it might be worth adding a "serial console" section to https://github.com/raspberrypi/documentation/blob/master/remote-access/README.md ? :man_shrugging:

green-green-avk, 2020-03-05:
It sounds funny but my initial intent to create the proposed client was the access to Raspberry-Pi through serial to use `raspi-config` and possibly other ncurses software. A long search for a suitable client brings nothing. Another concern was the requirement of a specific screen keyboard, so I just made it built-in.

duckida, 2020-04-14:
To me that looks like a terminal for the Droid and not an SSH Client

green-green-avk, 2020-04-15:
Yes, it also has a local pty support.

JamesH65, 2020-04-15:
> Although @JamesH65 do you think it might be worth adding a "serial console" section to https://github.com/raspberrypi/documentation/blob/master/remote-access/README.md ? 🤷‍♂️

Hmm, does appear to be a gap in the docs there. Might be worth spending some time on that. Not sure we want a list of third party serial terminals though, been trying to get rid of that sort of thing.

aallan, 2021-06-07:
Policy is no third party products. Closing.

closed 2021-06-07 by aallan