raspberrypi/documentation #1432: Windows ssh client

parlitu, 2020-03-13:
You can now connect with ssh using Windows Powershell 7
https://devblogs.microsoft.com/powershell/announcing-powershell-7-0/

lurch, 2020-03-13:
I don't use Windows myself - does this offer something that https://github.com/raspberrypi/documentation/blob/master/remote-access/ssh/windows.md or https://github.com/raspberrypi/documentation/blob/master/remote-access/ssh/windows10.md don't already cover?

parlitu, 2020-03-13:
Covered but outdated. The Documentation should be updated by removing Putty and include the new PowerShell 7 ssh mode...

closed 2020-03-13 by parlitu

lurch, 2020-03-13:
We always welcome Pull Requests... :slightly_smiling_face:

JamesH65, 2020-03-13:
Unless Powershell7 works on all current Windows version that are still in support, we should not be removing Putty.

Is the SSH in Powershell a graphical/windows one? Or command line?

parlitu, 2020-03-13:
What operating systems does PowerShell 7 support?
PowerShell 7 supports the following operating systems on x64, including:

Windows 7, 8.1, and 10
Windows Server 2008 R2, 2012, 2012 R2, 2016, and 2019
macOS 10.13+
Red Hat Enterprise Linux (RHEL) / CentOS 7+
Fedora 29+
Debian 9+
Ubuntu 16.04+
openSUSE 15+
Alpine Linux 3.8+
Additionally, we support ARM32 and ARM64 flavors of Debian and Ubuntu, as well as ARM64 Alpine Linux.

parlitu, 2020-03-13:
[image]