raspberrypi/documentation #1434: RealVNC is NOT free

TiberiusJ, 2020-03-14:
This goes against the whole open source concept. Damn organisations charging for software that costs as much as the computer one is trying to load it on. Its as bad as the Pi resellers charging double (or more) the cost of the board when the concept was a $25 computer or whatever rip-off equivalent these freeloaders join the band-wagon to rip off the consumer.

Please please please only recommend freeware/open source VNC/Remote desktop software that is also compatible with the computers that might be connecting to it. EG. Windoze, Linux variants and Mac OSX (Catalina please)

Please please please then, demonstrate an easy install script. Then newbies can attempt to learn about Pi and Linux easily from a comfortable, known surroundings.

JamesH65, 2020-03-14:
It's free on Raspberry Pi.

We will recommend what we think is suitable, whether it costs money or not.

Caveat Emptor. If you don't like people charging for stuff, don't buy it their stuff.

TiberiusJ, 2020-03-14:
No you have to activate it and to do that you have to create an account.
No way José

JamesH65, 2020-03-14:
Account creation costs nothing AFAIK, but if you don't want to use it, that is certainly your prerogative.

Free, as in beer.

TiberiusJ, 2020-03-14:
You have obviously never heard of technical security. Remote access for all
and sundry because uninformed blindly click yes. A cloud is someone elses
computer - you're free to access that kind of recource if you wish. I will
continue to be paid to advise organisations and professionals not to.

Each to his own.

JamesH65, 2020-03-14:
Strawman argument, but closing issue as we won't be making any changes in this are for the moment.

closed 2020-03-14 by JamesH65

TiberiusJ, 2020-03-14:
That's the best you can troll? Carry on and close for all you care, I am
still right.