raspberrypi/documentation #1439: Adding  telebit.io

ravish0007, 2020-03-18:
I would like to add one more way of accessing pi over the Internet by telebit.io which is created by @coolaj86

JamesH65, 2020-03-18:
On what page?

ravish0007, 2020-03-18:
This very page raspberrypi/documentation/blob/master/remote-access/access-over-Internet/README.md

JamesH65, 2020-03-18:
Hmm, got a lot on that page already, not sure we need any of them to be honest. Will check with comms teams.

lurch, 2020-03-18:
@JamesH65 See also #1133 and #1421 !

JamesH65, 2020-03-18:
Currently talking to the comms team to see what to do. My inclination is to get rid of that section completely, or just tell people to Google.

phiney0711, 2020-03-18:
Hi.  I work for "remote.it" that is also listed on this page.  We would hate to lose our link on this page because we've been there for 5 years and tens of thousands of pi users have downloaded and used our solution.  But I understand if that's best for the community.  May I suggest stop adding new solutions to the page instead of deleting it completely?  I agree there are too many at this point.  Thank you for your consideration.

ravish0007, 2020-03-18:
telebit.io is so incredible and also open source, It's totally free and costs no charges at all.
I got to know on that by randomly watching this video "Look Ma No Cloud" ( https://www.youtube.com/watch?v=E1Q2MWGCADo ), perhaps you might enjoy this :)

JamesH65, 2020-03-20:
It's been decided that since we cannot review or even list all possible online options we will remove all references to third party systems.

closed 2020-03-20 by JamesH65

ravish0007, 2020-03-20:
Sounds acceptable.

phiney0711, 2020-03-20:
We understand.  It had gotten a little overwhelming in the number of choices.  

Would you consider leaving it up a few more weeks?  The reason is with Corona Virus and the new "work-from-home" trend we are seeing overwhelming interest in setting up remote access via a Raspberry Pi to office LANs.  It seems to be helping a lot of small businesses as a low-cost solution to access their computers on the office LAN via a Raspberry Pi acting as a bridge (without the need to port forward).  Seems many small business do not have IT skills to port forward their gateway and port forwarding a terrible security issue too.  So a PI + remote.it gives full remote access to all devices/computers on office LAN without port forwarding.  Of course please go ahead and remove all the "solutions" on that page as planned whenever things settle down if that's acceptable.  Thanks you.

joshwget, 2020-03-20:
What if a "registry" of vendors were to be maintained somewhere else? Would you be willing to link to that if it was well-maintained @JamesH65?

Vendors like remote.it, Deviceplane, etc.. can maintain this registry and keep the burden off the Raspberry Pi Foundation.

JamesH65, 2020-03-20:
Wikipedia perhaps, or elinux? As long as it's found by Google. We don't really like linking to external sites as we cannot be held responsible for what they say, although people still blame us!