raspberrypi/documentation #1445: Disable RaspAP when rebooting

agoralive, 2020-03-25:
Hi

I have read a lot and found this: https://github.com/billz/raspap-webgui/issues/83

Is there another way the disable RaspAP (or just the Hotspot) when Rpi is rebooting?

I can stop the HotSpot, but I would like it to not run unless I activate it....

Perhaps it could "just" remember the setting, when rebooting?

JamesH65, 2020-03-25:
This does not appear to be a documentation issue, and RaspAP is nothing to do with us as far as I know.

closed 2020-03-25 by JamesH65

lurch, 2020-03-25:
@agoralive This repo is **only** about the official documentation at https://www.raspberrypi.org/documentation/

Any other questions should be directed to https://www.raspberrypi.org/forums/

noxware, 2022-10-16:
This worked for me:

```
sudo systemctl disable hostapd dnsmasq raspapd
```

If you don't disable the `raspapd`, it will reconfigure the other two on boot.