raspberrypi/documentation #1450: Mention the 'stable' firmware option in the docs

jwatt, 2020-03-29:
hardware/raspberrypi/booteeprom.md has a section [Beta firmware](https://github.com/raspberrypi/documentation/blob/master/hardware/raspberrypi/booteeprom.md#beta-firmware) which mentions setting `FIRMWARE_RELEASE_STATUS="beta"` to get more recent firmware. It does not make any mention currently of the `stable` value that can be used since [3be67ad8a869](https://github.com/raspberrypi/rpi-eeprom/commit/3be67ad8a8690e2661725a1eeacbddc1da464460). It probably should since most users should avoid the beta firmware.

JamesH65, 2020-03-30:
@pelwell @timg236 Since you guys are the experts, can someone knock up a quick PR?

timg236, 2020-03-30:
When I update the documentation next for the new features in the beta I'll do that, not this week though

lurch, 2020-03-30:
Should this be closed as a duplicate of #1436 ?

JamesH65, 2020-03-30:
Yes.

Duplicate of #1436

closed 2020-03-30 by JamesH65