raspberrypi/documentation #1451: Some Pi HDMI interfaces do not support audio

kylos101, 2020-03-29:
Hi,

I was thinking of adding [something like this](https://github.com/kylos101/documentation/commit/7cedd039af5c84e01a554f63f3fce669183b4dab) to audio-config.md.

My problem was that audio was not working on my Rasberry Pi 4 Model B over HDMI1. It never occurred to me that one HDMI interface would support audio, and the other would not. This calls that out upfront so that folks can do that investigation prior to making configuration changes.

Thanks,

Kyle

JamesH65, 2020-03-30:
Audio should work fine on both HDMI ports. Have you correctly assigned the audio destination using the Audio task bar icon? You should also update to all the latest software.

kylos101, 2020-03-30:
Hi @JamesH65 ,

Thanks for getting back to me so quickly! 👍 

Rasbian was updated when I installed my Rasberry Pi (it asks to do an update). Is that what you mean by update to all the latest software, or should I do it some other way?

I tried all of the things mentioned [in this article](https://www.raspberrypi.org/documentation/raspbian/updating.md). They are:
1. Desktop Volume Control (assert HDMI is selected, not Analog)
2. Command-line
3. raspi-config
3. Updating /boot/config.txt and rebooting

I only have one Rasberry Pi 4 Model B. Have you tested HDMI1 to see if it yields sound for you? I would be curious to see how this behaves for other users with similar hardware.

[Check out this forum post, there's a picture describing the same symptom I had, where one interfacet supports sound, but the other does not.](https://www.raspberrypi.org/forums/viewtopic.php?p=1550253)

JamesH65, 2020-03-30:
By default audio should just work on both HDMI ports. Right click on the speaker icon and select HDMI. Yes, this has been extensively tested.

To get the very latest software, start up a terminal windows

```
sudo apt update
sudo apt full-upgrade
```

kylos101, 2020-03-30:
I'll try the update commands you shared, plug into HDMI1, reboot, assert
HDMI is set in the speaker icon, and get back to you.

kylos101, 2020-03-30:
I edited my previous post, I had meant to link to audio-config.md in the first link, but was linking to an update article by accident.

Sadly there is no sound after getting the latest software, powering off, plugging into HDMI1, and starting back up.

Here is the output of the command-line method for audio configuration.

```
pi@pi-kitchen:~ $ amixer cset numid=3 2
numid=3,iface=MIXER,name='PCM Playback Route'
  ; type=INTEGER,access=rw------,values=1,min=0,max=3,step=0                                               
  : values=2

```

I'll upgrade a screenshot of the speaker icon next via VNC.

kylos101, 2020-03-30:
Hi @JamesH65 here is a screenshot showing HDMI is selected while connected to HDMI1. There is no audio while connected to HDMI1. 

[image]

Do you have a Pi 4 Model B to test HDMI1 with?

Take care,

Kyle

JamesH65, 2020-03-31:
> 
> Do you have a Pi 4 Model B to test HDMI1 with?
> 

Of course. What I don't have whilst working from home is a monitor with a speaker in it. 

@spl237 Do you have the time/equipment to quickly check that HDMI1 audio is still working correctly with the latest firmware?

lurch, 2020-03-31:
I wonder if using VNC is confusing the audio output mechanism somehow? (just a random guess) :man_shrugging: 

> What I don't have whilst working from home is a monitor with a speaker in it.

No HDMI TV in your lounge James?

spl237, 2020-03-31:
HDMI audio is definitely working on my current test image, which was updated to latest software about 4 weeks ago. Has firmware changed since then?

JamesH65, 2020-03-31:
> I wonder if using VNC is confusing the audio output mechanism somehow? (just a random guess) 🤷‍♂
> 
> > What I don't have whilst working from home is a monitor with a speaker in it.
> 
> No HDMI TV in your lounge James?

Yes, but with two children and a wife, its not like I up sticks and just use it.  

As for VNC, that hasn't been mentioned but is on that screen shot. What happens if it is turned off.

kylos101, 2020-04-04:
Just checking in, let me know if there's anything else I can do, or if you'd like me to repeat any tests. Take care, gang.

JamesH65, 2020-04-04:
I've not had a chance to check this, might give it a go early next week if I can transplant everything. However, I'm not expecting any problems - this has worked in the past and I don't think there have been any changes that might be causing problems.

kylos101, 2020-04-04:
Appreciate the help @JamesH65 !

As an FYI, [I've been using this small touchscreen w/speakers](https://www.amazon.com/Lebula-Touchscreen-Raspberry-1024X600-Capacitive/dp/B07VNX4ZWY). 

I'll try testing with a television, to see if the experience is any different.

spl237, 2020-04-07:
https://github.com/raspberrypi-ui/lxplug-volume/issues/7 looks to be the same thing

spl237, 2020-04-07:
Hi James,

Have a look at https://github.com/raspberrypi-ui/lxplug-volume/issues/7 <https://github.com/raspberrypi-ui/lxplug-volume/issues/7> - looks like PiTop have got some more info on the problem.

Thanks,

S.

m-roberts, 2020-04-07:
@kylos101 you need to run `amixer cset numid=3 3` to get audio out of HDMI1. This is currently not available through the UI.

lurch, 2020-04-07:
Also more detailed technical info in https://github.com/raspberrypi/firmware/issues/1243

JamesH65, 2020-04-29:
This appears to be an issue with the software not the documentation. Not sure what the current firmware state is with respect to this, but I would expect that in the nr future that HDMI should just work by default on both ports. @spl237 Any idea of the state of this?

kylos101, 2020-05-02:
Thank you  @m-roberts , @lurch , and @JamesH65  for sharing! I'm going to close this issue for now seeing as there are separate issues to fix the software. Let me know if you'd like me to reopen this and submit a documentation update.

closed 2020-05-02 by kylos101