raspberrypi/documentation #1461: UART documentation does not mention RPi4

rgov, 2020-04-08:
https://www.raspberrypi.org/documentation/configuration/uart.md says:

> By default, on Raspberry Pis equipped with the wireless/Bluetooth module (Raspberry Pi 3 and Raspberry Pi Zero W)

I believe the RPi4 does have a wireless/Bluetooth module, so this should be expanded to clarify.

rgov, 2020-04-08:
Generally I find this document very confusing. I've had to read it many times. I'll probably contribute a revision that simplifies it somewhat.

One point I'm confused about is if `enable_uart=1` then the core frequency is fixed to 250MHz so the Mini UART can have a constant baud rate.

But then it says if Mini UART is not the primary UART, you must add `core_freq=250`... to fix the core frequency to 250 MHz so that the Mini UART can have a constant baud rate?

----

The following table summarizes the default configuration of the UARTs.

| Model               | mini UART            | PL011                 |
| ------------------- | -------------------- | --------------------- |
| Raspberry Pi 1      | secondary            | primary               |
| Raspberry Pi 2      | secondary            | primary               |
| Raspberry Pi 3      | primary (disabled)   | secondary (Bluetooth) |
| Raspberry Pi 4      | primary (disabled)   | secondary (Bluetooth) | 
| Raspberry Pi Zero   | secondary            | primary               |
| Raspberry Pi Zero W | primary (disabled)   | secondary (Bluetooth) |
 
In Linux device terms:

| Linux device   | Description                  |
| -------------- | ---------------------------- |
| `/dev/ttyS0`   | mini UART                    |
| `/dev/ttyAMA0` | PL011                        |
| `/dev/serial0` | primary UART (Linux console) |
| `/dev/serial1` | secondary UART               |

JamesH65, 2020-04-09:
Looks like a number of issues on that page, since its was written before the Pi4 came out. That has different core frequencies etc, and I think more UARTS anyway. I'll need to check what we need to add on this. Thanks for the report.

lurch, 2020-04-09:
IIRC the 'primary' UART (the one symlinked to `/dev/serial0`) is also the one that appears on GPIOs 14 and 15 (pins 8 and 10) on the Pi's 40-pin header, if you've enabled serial console in `raspi-config`.
And yes, the Pi4 has 6 UARTs in total (one mini-UART and 5 PL011s).

rgov, 2020-04-09:
Glad to see the page get some attention.

I think one thing that makes it so confusing is that there are many different ways to refer to the UARTs, like

* primary vs secondary
* /dev/ttyS0 vs /dev/ttyAMA0
* /dev/serial0 vs /dev/serial1
* mini UART vs PL011
* console vs Bluetooth chip

I recognize they're not equivalent but the cognitive load on the reader's part is much harder due to having to translate between them. I encourage you to take the tables in my post above, adjusted for RPi4 as needed, as a quick reference that saves a few paragraphs of reading.

Another confusing part is that the "primary" device is (on some models) disabled by default, while the "secondary" remains enabled. That seems like a semantic contradiction, so some clarification would be good.

lurch, 2020-04-09:
...and there's also the "UART0" / "UART1" names used in the [datasheets](https://www.raspberrypi.org/documentation/hardware/raspberrypi/datasheets.md) and (indirectly) by [raspi-gpio](https://github.com/RPi-Distro/raspi-gpio) .

@pelwell Can confirm for sure, but the way that I understand things...
At the low-level there's:
* UART0 = TXD0/RXD0 in raspi-gpio = PL011 = /dev/ttyAMA0
* UART1 = TXD1/RXD1 in raspi-gpio = mini UART  = /dev/ttyS0

And then at a "higher" level there's:
* /dev/serial0 = primary = symlink to either /dev/ttyAMA0 or /dev/ttyS0 (depending on the model of Pi) = the serial console on the Pi's GPIO pins (if enabled)
* /dev/serial1 = secondary = symlink to either /dev/ttyS0 or /dev/ttyAMA0 (depending on the model of Pi) = the connection to the Bluetooth chip (if the Pi has one)
* And then I think some of the [device-tree overlays](https://github.com/raspberrypi/firmware/blob/master/boot/overlays/README) can also swap-over or disable primary or secondary?

On the Pi4, there's also UART2 -> UART5 which are all PL011. I've not played with them since the Pi4 launch, but IIRC they appeared in Linux as /dev/ttyAMAX (where the X at the end is a unique number).

pelwell, 2020-04-09:
@lurch your understanding of the UARTs is correct. The overlays in question are `disable-bt` which makes PL011 the primary and disables ttyS0, and `miniuart-bt` which again makes PL011 the primary but hooks ttyS0 up to the Bluetooth modem.

@rgov I like the table - it is a helpful summary which may be all that many people need. The additional UARTs on Pi4 could get a one-line mention, but they don't get any special treatment from the firmware or Linux - just a unique device name, as @lurch said.

andrum99, 2020-04-29:
> @rgov I like the table - it is a helpful summary which may be all that many people need. The additional UARTs on Pi4 could get a one-line mention, but they don't get any special treatment from the firmware or Linux - just a unique device name, as @lurch said.

Yes, the table would be a very useful addition. I'm quite familiar with the text, having worked on it previously, so would be happy to have a look at this.

JamesH65, 2020-11-19:
Fixed by #1560

closed 2020-11-19 by JamesH65