raspberrypi/documentation #1463: GPIO output voltage levels

szilvasyz, 2020-04-10:
Hello,
All places in the page /hardware/raspberrypi/gpio/gpio_pads_control.md there are mistakenly mentioned V(IH) and V(IL) parameters and their represented muneric values as the datasheet references for the output voltage of a pad. Instead, the datasheet parameters V(OL) and V(OH) reflect for output levels of pads. Please, correct these on the documentation page and the regarding voltage levels too. Thank you.
Zoltan

JamesH65, 2020-04-10:
@jimbojr @mike-stimson I thought that page was correct, any comment?

https://www.raspberrypi.org/documentation/hardware/raspberrypi/gpio/README.md

mike-stimson, 2020-04-10:
Looks ok to me. Can't see the problem.

Mike

szilvasyz, 2020-04-10:
Please take a look at the "subpage" linked by "Output [drive strength](https://www.raspberrypi.org/documentation/hardware/raspberrypi/gpio/gpio_pads_control.md)". That page, discussing characteristics of **output mode pads** (usable output currents and voltage levels) always refers to V<sub>IL</sub> and V<sub>IH</sub> **input mode parameters** of datasheet. But, in real, these parameters reflect the behavior of an input mode pad. For output mode you should use V<sub>OL</sub> and V<sub>OH</sub> parameters.

Moreover, not only the parameter names, but the parameter numeric values themselves are mixed up in the "drive strength" page. For example a 3.3V output mode pad should never go up to 0.9V (suggested by V<sub>IL</sub>) in normal circumstances (below the configured output current), but it must remain not more than 0.14V as V<sub>OL</sub> shows.

Of course, V<sub>IL</sub> and V<sub>IH</sub> are describing voltage levels of an input pad, which are to be interpreted as "low" and "high" logic levels, respectively. The quoted table from the datasheet on the main "GPIO" page is correct with absolutely usual values for such kind of I/O pad.

mike-stimson, 2020-04-16:
Ah, I see your point. @JamesH65 can you fix this? IIRC, I think this is because the very first BRCM datasheet had the output and inputs parameters reversed, and this explanation is probably based on that.

JamesH65, 2020-04-16:
Is it just this bit that needs reversing?

> VIL, the maximum low-level voltage (0.9V at 3V3 VDD IO)
VIH, the minimum high-level voltage (1.6V at 3V3 VDD IO)

JamesH65, 2020-04-16:
Ah, sorry, read this completely wrong.  It would be easier if someone who knows about this simply provides a PR!

JamesH65, 2020-04-16:
Or is it on that page that all IH/IL simply need to be changed to OH/OL?

mike-stimson, 2020-04-16:
Not quite, unfortunately. As Zoltan describes, the values of 0.9V and 1.6V
are not appropriate (they are the actual VIL/IH levels). Infact, the worked
example of a 16mA drive strength afterwards isn't quite right, as we only
define our VOH/OL levels with a 8mA drive setting, but whilst sink/sourcing
2mA (if the notes on the table are to be believed).

JamesH65, 2020-04-29:
In which case I have no idea what to change because I am a HW idiot. Going to need someone who knows more about this than me to update that stuff.

ernestguillem, 2021-08-15:
Sent a PR with the fix.
`https://github.com/raspberrypi/documentation/pull/2081`

commit 84b04667 (2021-08-31): Fixes #1463

Changed the VOH and VIH labels and values to reflect the correct ones from gpio-on-raspberry-pi.adoc
commit e2be977b (2021-08-31): Merge pull request #2081 from ernestguillem/fix_1463

Fixes #1463
closed 2021-08-31 by aallan