raspberrypi/documentation #1464: We are still waiting for Boot EEPROM hardware write protection information

McuMirror, 2020-04-11:
".. it is possible to physically write-protect both EEPROMs via a simple resistor change on the board. Details will be published in the schematics." -- Documentation

Not yet indicated in the schematics. Please add this information.
We'll have to wait for the Raspberry Pi org to release the EEPROM part of the schematics or simply tell us which trace/resistor to cut/remove. 

sorry for the new issue but i think this subject is forgotten.

JamesH65, 2020-04-11:
@jimbojr @mike-stimson Is this your domain?

lurch, 2020-04-14:
Closing as a duplicate of #1302

closed 2020-04-14 by lurch

McuMirror, 2020-04-14:
OK. Close this issue, but a simple explanation should not take weeks/months.

JamesH65, 2020-04-14:
Please use the linked issue for any comments, this one is closed.