raspberrypi/documentation #1470: Remote-access/VNC/README.md - VNC comes builtin to newer Raspbian Distros

duckida, 2020-04-14:
I was lookin through VNC and I saw you have to install it with a command. On my Pi, VNC is pre-installed.

lurch, 2020-04-15:
[The page](https://www.raspberrypi.org/documentation/remote-access/vnc/README.md) _does_ already say "VNC Connect from RealVNC is included with Raspbian.".
And the "install instructions" you mention are actually prefaced with "On your Raspberry Pi, run the following commands to make sure you have the latest version of VNC Connect:".

@spl237 @XECDesign Do you think it's worth tweaking the documentation to make it clearer that this is an optional step?

duckida, 2020-04-15:
Yes

spl237, 2020-04-15:
It's included in the full image; it's not in the standard image, but is available in Recommended Software; no need to use apt to install it.

It's probably worth updating the page to reflect that rather than suggesting the use of apt.

JamesH65, 2020-04-15:
See https://github.com/raspberrypi/documentation/pull/1476

lurch, 2020-04-15:
Fixed by #1476

closed 2020-04-15 by lurch