raspberrypi/documentation #1471: HARDWARE/Sense-Hat/README x Does not provide instructions on how to use the SenseHat

duckida, 2020-04-14:
I was browsing through  and found that the SenseHat docs do not tell you how to control it but only debug it.

lurch, 2020-04-15:
https://www.raspberrypi.org/documentation/hardware/sense-hat/README.md includes "After installation, example code can be found ..." ?

Maybe you're looking also for https://projects.raspberrypi.org/en/projects?hardware%5B%5D=sense-hat  :question:

duckida, 2020-04-15:
I think that noobs will think to come here but then just find this debug stuff

JamesH65, 2020-04-29:
Ouch, that page is pretty bad. Not sure what to do with it though. Will think about it.

duckida, 2020-04-30:
We should make it a page on How To use sense hat

aallan, 2021-08-13:
Page now lives at https://www.raspberrypi.org/documentation/accessories/sense-hat.html, source for that now lives at https://github.com/raspberrypi/documentation/blob/develop/documentation/asciidoc/accessories/sense-hat.adoc.

Does someone that's used a Sense HAT want to offer up a PR for this one?

closed 2021-10-24 by github-actions[bot]