raspberrypi/documentation #1478: Document enable_jtag_gpio setting for config.txt

czak, 2020-04-19:
There seems to be an `enable_jtag_gpio=1` setting for `config.txt` which I've seen mentioned in a few blog posts and Github issues but not documented here.

I guess this is just a shortcut for `gpio=22-27=a4`, but maybe there's more to this and worth documenting?

I had something like this in mind: https://github.com/czak/documentation/pull/1/files. Let me know if this sounds OK and I can post that PR here.

References:
* https://github.com/raspberrypi/firmware/issues/639
* https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2835/BCM2835-ARM-Peripherals.pdf (section 6.2 - Alternative Function Assignments)

JamesH65, 2020-04-19:
I have no idea about that option, but it's much easier if you just send in the PR rather than for us to go and find it somewhere else

czak, 2020-04-19:
> I have no idea about that option, but it's much easier if you just send in the PR rather than for us to go and find it somewhere else

:+1: closing in favor of #1479

closed 2020-04-19 by czak