raspberrypi/documentation #1492: Audio setup documentation does not mention bluetooth speakers

petercwallis, 2020-04-27:
and doesn't even acknowledge such things exist which may have been true 6 years ago.  The obvious thing to do is the connect via the blue tooth icon menu, (which can find the device) but gives an error saying to use the blue tooth option on the audio menu.  That naturally does not exist.

A google for the problem returns all sorts of solutions from three years ago including installing non existent packages with apt-get.  I get the impression that connecting blue tooth speakers is easy these days (the problem is solved) but yet to be documented?  Here would seem to be a sensible place; could someone who knows "the right" solution put it in here.

JamesH65, 2020-04-27:
A link to the page you think needs updating would be good.

XECDesign, 2020-04-27:
> but gives an error saying to use the blue tooth option on the audio menu. That naturally does not exist.

I was with you up until that bit. It's not an error, it's telling you to select the bluetooth device as the output in the audio applet. I'm not sure what you mean by saying it does not exist.

lurch, 2020-04-27:
ping @spl237 as I believe this is his bag?

spl237, 2020-04-27:
Bluetooth audio works - I've been using it for the last few days. As XECDesign points out, the message telling you to use the Bluetooth option in the audio menu is not an error; it is an instruction to select the Bluetooth device in the audio selector menu which is obtained by right-clicking on the volume icon immediately to the right of the Bluetooth icon.

I'll happily look into any bugs, but I am not aware of any...

lurch, 2020-04-27:
Glad to hear (no pun intended!) that it's working, but does the OP's confusion suggest that we need some better documentation (as he suggested) ?  I have no Bluetooth devices myself...

petercwallis, 2020-04-29:
My goodness! What a responsive crew you are!
So the up date is that it does work;  The reason I was being told to
connect via the audio menu was that the blue tooth side thought (sorry for
the anthropomorphism) that the head set was already connected (which it
presumably was) but it had not registered with alsa??

My 13 year old disconnected it from the bluetooth menu, then refound it on
the bluetooth menu, and it then appeared on the audio menu.

Just tried it again on a different rpi install and that works.  I have no
idea how it came to be that it had paired with blue tooth but not appeared
in the audio menu. perhaps it is simply a matter of changing the message to
something about the device is connected. If it is not an option in the
audio menu, disconnect it here and start again.?

P

JamesH65, 2020-11-19:
@spl237 I think we can close this issue, but are there any improvements to the message described that might help?

spl237, 2020-11-19:
Not that I can think of....

petercwallis, 2020-11-19:
Is it possible that someone can connect an audio device to the bluetooth
but not to alsa?  If not, yep close it - no idea what I did wrong.  If it
is possible, a) is there an auto check that can be done with a message on
how to fix it. b) if not is there a manual check described somewhere that
would check for such a configuration that should be added to or replace my
post.

spl237, 2020-11-19:
To be honest, this is all going to be academic once we are using PulseAudio, and that is imminent, so this issue will go away of its own accord at that point; no point leaving it open.

lurch, 2021-07-10:
> this is all going to be academic once we are using PulseAudio, and that is imminent, so this issue will go away of its own accord at that point; no point leaving it open.

Did that happen yet @spl237 ? (i.e. can this issue be closed?)

spl237, 2021-07-10:
Yes, PulseAudio is now the default - this can be closed

closed 2021-07-10 by lurch