raspberrypi/documentation #1495: 全新购买的树莓派4b刷入最新版系统raspberry后wifi能搜索到但是输入密码无反应

15230745073, 2020-04-28:
全新购买的树莓派4b刷入最新版系统raspberry后wifi能搜索到但是输入密码无反应
wpa工具也无效

lurch, 2020-04-28:
Google translates that to:
"After the newly purchased Raspberry Pi 4b is flashed into the latest version of the system raspberry, wifi can be searched but the password is not responded
wpa tool is also invalid"

That's a question for the [forums](https://www.raspberrypi.org/forums/) rather than this documentation repo.

closed 2020-04-28 by lurch

JamesH65, 2020-04-28:
Unfortunately, we can only deal with English language reports on here. Having used Google translate however, it seems you need to use the forums for help. This repo is for reporting problems with  the documentation.  www.raspberrypi.org/forums

15230745073, 2020-04-28:
Thanks, I will go to the forum for help