raspberrypi/documentation #1505: scripting.md page is weak

JamesH65, 2020-05-01:
This page could do with some extra stuff, even if only an offsite link to more examples.

JamesH65, 2020-05-01:
https://www.raspberrypi.org/documentation/linux/usage/scripting.md

andrum99, 2020-05-01:
What about a link to some MagPi articles? Is it possible to just add the MagPi articles to the website, if they're not already there somewhere? I mean as text, not just PDFs.

andrum99, 2020-05-03:
Pimoroni "Hello World" in bash: https://learn.pimoroni.com/tutorial/bash/getting-started-with-bash

Core Electronics: https://core-electronics.com.au/tutorials/raspberry-pi-workshop-for-beginners.html#ch3

Adafruit: https://learn.adafruit.com/an-illustrated-guide-to-shell-magic-typing-less-and-doing-more/write-shell-scripts

I particularly like the Adafruit one, since it has good links to examples of real world usage of bash scripts, and at the bottom of the page there are links to two reference manuals. Pimoroni's page is the simplest of the three. Core Electronics includes video tutorials.

aallan, 2021-06-07:
Closing in light of #1911.

closed 2021-06-07 by aallan