raspberrypi/documentation #1509: SELF_UPDATE=1 disappears after every update

starbasessd, 2020-05-02:
Trying to have Pi4 4GB do auto eeprom updates due to being net-booted
eeprom  beta  pieeprom-2020-04-16.bin
pi@raspi1:~ $ vcgencmd bootloader_config
[all]
BOOT_UART=0
WAKE_ON_GPIO=1
POWER_OFF_ON_HALT=0
DHCP_TIMEOUT=45000
DHCP_REQ_TIMEOUT=4000
TFTP_FILE_TIMEOUT=30000
TFTP_IP=
TFTP_PREFIX=0
BOOT_ORDER=0x21
SD_BOOT_MAX_RETRIES=3
NET_BOOT_MAX_RETRIES=5
[none]
FREEZE_VERSION=0
Follow instructions to extract, modify, re-apply, and re-install here:
https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2711_bootloader_config.md
adding SELF_UPDATE=1 after the line [all], and before and after [none]
I have been able to have BOOT_ORDER=0x21 to do Net Boot, successfully.
Just can't get the SELF_UPDATE to stay.

lurch, 2020-05-04:
This is probably an issue for https://github.com/raspberrypi/rpi-eeprom/issues rather than the documentation repo?

starbasessd, 2020-05-04:
My bad. I was thinking maybe it was documented incorrectly. Will take it there.

closed 2020-05-04 by starbasessd