raspberrypi/documentation #1510: How do I measure the voltage of my PI 4 B?

hackercyclops12, 2020-05-03:
How do I measure the voltage using commands via the command line?

Thanks in advance :)

lurch, 2020-05-04:
https://github.com/raspberrypi/documentation/blob/master/raspbian/applications/vcgencmd.md#measure_volts-block

However, if you mean the power-input voltage, then note that if it's too low, a [warning icon](https://github.com/raspberrypi/documentation/blob/master/configuration/warning-icons.md) will be displayed.

I suspect that this is really the sort of question you should be asking on the [forums](https://www.raspberrypi.org/forums/) though, so closing this issue for now. Feel free to follow up with further comments if you have specific questions about the [documentation](https://www.raspberrypi.org/documentation).

closed 2020-05-04 by lurch

hackercyclops12, 2020-05-04:
Thankyou that really helped me :)