raspberrypi/documentation #1512: add rpi-update page

JamesH65, 2020-05-03:
A quick grep and apparently there are no instruction on how to use rpi-update. Although it's only recommended to use when told to, we really ought to have some sort of instructions available, especially on how to revert it effects.

lurch, 2020-05-04:
I realise they're embedded in other pages, so probably not what you were looking for, but there's info at https://github.com/raspberrypi/documentation/blob/master/linux/kernel/updating.md#manual-updates-using-rpi-update and https://github.com/raspberrypi/documentation/blob/master/hardware/display/troubleshooting.md#have-you-updated-raspbian (which I guess should probably be updated if you add a separate rpi-update page?)

JamesH65, 2020-05-05:
Fixed by #1526

JamesH65, 2020-05-05:
I still need to adjust the other pages to take this one in to account.

JamesH65, 2020-05-05:
OK, have changes the other pages.

closed 2020-05-14 by JamesH65