raspberrypi/documentation #1520: Document `force_eeprom_read` option

hvenev, 2020-05-04:
A `config.txt` option names `force_eeprom_read` is being mentioned around the Internet. What does it do? I believe it should be documented.

lurch, 2020-05-04:
ping @timg236 @pelwell

pelwell, 2020-05-04:
`force_eeprom_read` does exactly what those mentions on the internet suggest - it controls whether or not the firmware reads the HAT EEPROM. By default, the firmware tries to read the EEPROM if the dt-blob file (an external dt-blob.bin, or the built-in one) has an ID_SDA declaration, which is normally true for all boards with a 40-pin header. You can force it to try by setting `force_eeprom_read=1`, but that is unlikely to work without the ID_SDA and ID_SCL declarations. It's more likely that somebody may want to disable the HAT EEPROM by adding `force_eeprom_read=0`.

lurch, 2020-05-04:
Ahh, so this is actually a ternary option? :wink: (0, 1 and unset)

pelwell, 2020-05-04:
I'll assume that's a joke and that I don't need to explain how a default value is part of the (in this case binary) set, not some new value.

lurch, 2020-05-05:
Fixed by #1521 :slightly_smiling_face:

closed 2020-05-05 by lurch