raspberrypi/documentation #1522: BCM2711 EEPROM config question

lurch, 2020-05-04:
I just noticed that when I run `vcgencmd bootloader_config` (as documented at https://github.com/raspberrypi/documentation/blob/master/hardware/raspberrypi/bcm2711_bootloader_config.md ) on my Pi4B the output includes `[all]` and `[none]` lines. (The same output is present in the result of `rpi-eeprom-config /lib/firmware/raspberrypi/bootloader/critical/pieeprom-2020-04-16.bin` so I guess these are actually values in the EEPROM config itself rather than being something added by `vcgencmd`)

What do these lines mean @timg236 ? Should they be added to the [documentation](https://github.com/raspberrypi/documentation/blob/master/hardware/raspberrypi/bcm2711_bootloader_config.md) ?

timg236, 2020-05-04:
It already is documented https://www.raspberrypi.org/documentation/configuration/config-txt/conditional.md

lurch, 2020-05-05:
Fixed by #1523 , thanks Tim.

closed 2020-05-05 by lurch