raspberrypi/documentation #1530: Pi4 Model B 4GB Screen Goes Black for 2 to 4 seconds at a time

AlGantori, 2020-05-06:
- I purchased the Pi4 Model B 4GB 
- Brand new wireless Logitech wireless Mouse/Keyboard
- Right from the start after using the Pi Imager 1.2 and booting for the 1st time on the living room TV either I get **no signal** or **screen goes black**.
- Initially I had tried **hdmi_safe=1** but soon enough realized that did not work, next I realized it was ignoring my steps to try different specific modes.
- I tried another TV same problem so I realized the Pi is the problem.
- I tried 5G wireless on my router not a solution
- To try a different **hdmi_mode** I have to extract microSD bring it to PC, mount it, edit config.txt, go back, mount it, re-power the Pi,  it was a pain last night.
- Even if I wanted to edit on the PI itself with keyboard and linux editor, the screen was going black every second.
- the Pi hdmi output is still very flaky, the screen flashes, goes black for 2-4 seconds, **on it's own**, and also on mouse clicks, I have tried several settings in the readable partition **config.txt** file. 
- I think I am currently using hdmi_group=1 (CEA) hdmi_mode = 78
[image]

- My TV specs are the following

 1. TV VIZIO model E701I-A3
 2. Refresh rate  120 Hz 
 3. Resolution 1920 x 1080
 4. Image Aspect Ratio 16:9
- I have NOT experimented with **hdmi_group=2** (DMT) 
- I am tired doing guess work
I feel this is a terrible product, unless this board is buggy somehow?

JamesH65, 2020-05-06:
Have you changed the HDMI cable itself - recent forum reports have similar symptoms that were fixed by changing the cable. Also try the other HDMI port just in case the plug is defective.

It's not a terrible product. If every one of the more than 2 million or so sold so far was having HDMI problems I am sure we would have noticed.

lurch, 2020-05-06:
Try the troubleshooting forum https://www.raspberrypi.org/forums/viewforum.php?f=28

AlGantori, 2020-05-12:
I am using the 6ft version of this UGREEN brand, it looks well made, bought here

https://www.amazon.com/UGREEN-Adapter-Ethernet-Compatible-Raspberry/dp/B06WWQ7KLV/ref=sr_1_3?dchild=1&keywords=raspberry+pi+4+hdmi+cable&qid=1589274571&s=electronics&sr=1-3

Is there a length max limitation, I noticed lots of kits come with a foot cable.  I am not ready yet to tuck this Pi behind the TV and perhaps never.  Do I have to buy the 3ft version?

The video output is running for longer time segments but it still blanks out for 1 to 2 seconds and comes back every one in a while.  This happens without touching the cables, mouse or keyboard.   

Every 3rd or 4th power-on run, it will loose video signal after like 10 minutes or so. TV will say no signal.  I move the mouse or press on keyboard but I get nothing.   At which point I have to re-power Pi.  I will eventually look into having a soft-reset button to initiate a controlled shutdown, since this board does not come with one.

It's not clear to me if this loss of video is related to the Pi "blanking" feature?

Today I tried setting up an autostartup to launch chrome in kiosk mode.  I noticed the the suggested `@xset s noblank`  which I thought might fix the problem.
I think it might have lost video since that edit but I will have to report later once I comfirm it has done it few times.

```
@lxpanel --profile LXDE-pi
@pcmanfm --desktop --profile LXDE-pi
@xscreensaver -no-splash

@xset s 0 0
@xset s noblank
@xset s noexpose
@xset dpms 0 0 0

chromium-browser --kiosk /home/pi/Desktop/MyHome.html
```

I have searched the suggested forum link and seen open/unresolved comparable issues like:

https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=272168&p=1650365&hilit=blank#p1650365 

https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=266861&p=1649929&hilit=blank#p1649929

I will search some more.  But seriously I feel these kind of loose forums is a waste of people time, where the questions and content are un-moderated, no participants rating, so you end up with just garbage talk.

A question like "RPi4 display problem" can be a Zillion things.

JamesH65, 2020-05-12:
This is clearly not a documentation issue, but a technical support issue. Whether you like the forum or not, the forum is where technical issues are sorted out. It's worked well for 8 years, and they are generally regarded as the best place to get Raspberry Pi information. We don't do up or downvoting. 

Note that the forum IS moderated, and is the only place where actual Raspberry Pi engineers answer questions (except for the Github issue trackers, but they also are not for technical support, but for, rather obviously, issue tracking)

lurch, 2020-05-12:
I agree with James that this is a problem for discussion on the forums, not on the documentation repo.
But have you tried experimenting with the "Screen Blanking" option in the Raspberry Pi Configuration program?

closed 2020-05-12 by lurch

AlGantori, 2020-05-16:
Just in case some poor soul lands here,... but I will be posting to the forum as well...

[image]

 It looks like I will have to return the first Pi.

hackercyclops12, 2020-05-17:
Edit the config.txt file and add 

Config_hdmi_boost=11

This should do the trick

JamesH65, 2020-05-17:
> Edit the config.txt file and add
> 
> Config_hdmi_boost=11
> 
> This should do the trick

Not on a Pi4, that has no effect. See https://www.raspberrypi.org/documentation/configuration/config-txt/video.md

hackercyclops12, 2020-05-18:
How long is the HDMI cable you are using?

AlGantori, 2020-05-18:
The standard length 18 meters that is. No I mean, 1.8 meters, I forgot the decimal/point system, I grew up with, now I am somewhat used to the bloody british system in the US.

6 feet should kosher, from the start I was not really paying attention to the comments about cables, I see too many times the cable and ps thingy being used as an escape goat.

I just happened to have a sour pi for my startup.   
Basically it's a bad board, point final.  It slipped thru the pi makers testing let them deal with it.

Config_hdmi_boost=11 sounds like it was going to cause a fire in my living room.
I did try that in my very first night with that pi, and discovered too soon it did not apply for my pi4, that actually pissed me off so much junk documentation about config.txt ZILLION possibility settings.
I guess that are still living in the dark ages in Britain, with TVs having a wooden frames around, them LOL.

lurch, 2020-05-18:
At least you're not stuck with the Raspberry Pi outputting 640x480 VGA resolution :stuck_out_tongue_winking_eye: :tv: 
IMHO flexibility is a good thing - most of the time the Pi picks up the correct settings automatically, but for the few occasions it doesn't you can tweak it manually via `config.txt`.

hackercyclops12, 2020-05-19:
Adding hdmi_safe = 1 
Setting hdmi_safe to 1 will lead to "safe mode" settings being used to try to boot with maximum HDMI compatibility. This is the same as setting the following parameters:

hdmi_force_hotplug=1
hdmi_ignore_edid=0xa5000080
config_hdmi_boost=4
hdmi_group=2
hdmi_mode=4
disable_overscan=0
overscan_left=24
overscan_right=24

Source:
https://www.raspberrypi.org/documentation/configuration/config-txt/video.md

JamesH65, 2020-05-19:
Note that hdmi_boost is ignore on the Pi4. 

**config_hdmi_boost**
Configures the signal strength of the HDMI interface. The minimum value is 0 and the maximum is 11.

The default value for the original Model B and A is 2. The default value for the Model B+ and all later models is 5.

If you are seeing HDMI issues (speckling, interference) then try 7. Very long HDMI cables may need up to 11, but values this high should not be used unless absolutely necessary.

**This option is ignored on the Raspberry Pi 4.**

AlGantori, 2020-05-20:
The best way to handle documentation and give less headaches for your users (customers) is to dedicated the documentation to each product.
If you want to have common sections that apply to several products, then that will be your challenge on how to achieve that via some "includes" / documentation build process behind the scenes.

Having dedicated documentation for Pi3 vs Pi4 
1. Removes clutter
1. Removes the unnecessary need to say "it does not apply here or there".
2. Allow independent revisions.
3. ...

The moment your create a new product your fork its documentation.

lurch, 2020-05-20:
The **vast majority** of the documentation applies equally well to all models of Raspberry Pi - the engineers at Raspberry Pi go to great efforts to make all the Raspberry Pi models work as similarly as possible.
This is why for instance there's only a single "Raspbian" download on https://www.raspberrypi.org/downloads/raspbian/ and not separate Raspbian downloads for Pi1, Pi Zero, Pi 2, Pi3, Pi4. And the Pi1 is still supported even though it was released 8 years ago (newer Pis are obviously much faster).

There's only a _few_ exceptions where things work slightly differently, such as the low-level `config.txt` details you're encountering here.

All IMHO of course - I should clarify that these are only my personal opinions!

aallan, 2020-05-20:
There would be almost no variation between documentation for the Raspberry Pi 3 and the Raspberry Pi 4, or the original Raspberry Pi and the latest come to that. Forking the documentation on a per-product basis is *not* the best way to support the user base at this time.