raspberrypi/documentation #1531: Pi4 Model B 4GB How to get rid of the TWO inches black border all around the TV screen?

AlGantori, 2020-05-06:
- During the 1st boot, Pi asked about with the video output was filling all of the TV screen, I had to ignore that because the video signal would go black right after password settings or selecting the wireless network, it took like 10 to 15 fails before I would go past the network selection.

- Besides the screen going BLACK randomly and on mouse clicks, 
I still 2 inches black border all around the TV screen

- Where is the setting to fill the TV screen?

JamesH65, 2020-05-06:
Not a documentation issue, please use the forums.

closed 2020-05-06 by JamesH65

lurch, 2020-05-06:
> Besides the screen going BLACK randomly and on mouse clicks

Maybe you need a better HDMI cable and / or better PSU?

> I still 2 inches black border all around the TV screen

Sounds like you need to fiddle with the overscan settings https://github.com/raspberrypi/documentation/blob/master/configuration/config-txt/video.md

AlGantori, 2020-05-09:
The actual link is https://github.com/raspberrypi/documentation/blob/master/configuration/config-txt/video.md#disable_overscan

disable_overscan=1   fixed the problem for me and got rid of the 2 inches black UNUSED border.

In simple English it means enlarge the HDMI OUTPUT to the TV screen, that's the English the documentation needs to use.

- disable_overscan=1  means **enlarge** the video OUTPUT to the TV/Monitor.
- disable_overscan=0 mean **shrink** the video OUTPUT to the TV/Monitor.
- The default value of disable_overscan is 0.

Something like the above would have avoided posting an issue in the first place.

In most fields I see this "garbage" lingo leaking from the engineers to the end-users.  My grandma and myself included, DO NOT want to know about what "overscan" really means.  When I plugged my Chromecast, AppleTV, ... last year on this TV the output occupies the whole screen that's what I expect, not have having to fool around some config.txt

Even the name of this setting/variable is backwards, "disable" is a "negative" so you need a double-negative to mean "enable"

Mining the config.txt documentation web page reveals we have the following settings using "disable"

- sdtv_disable_colourburst
- disable_touchscreen
- disable_overscan
- disable_fw_kms_setup

and the following settings using "enable"
- enable_tvout-pi-4b-only
- hdmi_enable_4kp60-pi-4b-only
- enable_dpi_lcd

make up your mind, bloody Pi, the **logical** one to use with 0 and 1 is "enable"

If you love "disable" then the English for the following setting should read "Disable/enable the touchscreen"

[image]

I am sorry to say that my experience with the Pi has been a sour one, no sorry, not a lemon-meringue one.

Some of the problems stem from this poorly written documentation.

Sorry but I will NOT post anything to the forum, I am against forums and learning new interactions and mostly limitations of each forums' software.  
This kind of wheel has been already invented as we already have StackOverflow and its parent site to track HW/SW/... issues, with moderators, up/down answer voting, notifications, etc...

JamesH65, 2020-05-10:
The forum will be eternally grateful.

Overscan is the terminology used by the TV makers (moist of them - some use different terms).  We did not invent it. We need to 'turn off' the overscan implemented by the TV. Disable means turn off.

Not only that but people have been getting on fine with this terminology since it was added to the very first Pi 8 years ago. You are the first person to complain. So it seems to me that we cannot have got it so wrong. 

So sorry, your argument is not convincing.

lurch, 2020-05-10:
Also, if the overscan was turned off by default, then it would mean that users who have (older?) TVs that **do** have overscan, would have the edges of the display missing (e.g. the menu buttons probably wouldn't be visible), and I imagine that'd be a _much_ more frustrating experience than having "unused borders" on your display.