raspberrypi/documentation #1544: Document how to use multiple Raspberry Pi boards with a single keyboard and mouse

aallan, 2020-05-14:
There is an open source version of [Synergy](https://symless.com/synergy) which allows you to share a single keyboard and mouse between multiple computers. See https://www.rauchland.com/articles/using-barrier-to-control-multiple-pcs-with-one-keyboard-and-mouse/ for an existing walkthrough (that isn't Raspberry Pi specific).

lurch, 2020-05-14:
With the obvious caveat that you need a separate display attached to each Raspberry Pi :wink:

andrum99, 2020-05-23:
See Gordon's rather excellent write up and video on the blog at https://www.raspberrypi.org/blog/share-your-keyboard-and-mouse-between-computers-with-barrier/

lurch, 2020-05-23:
@aallan and @ghollingworth actually work together (although not in the same office ATM, for obvious reasons) :smiley:

aallan, 2020-06-04:
I think I opened this before Gordon's article got published based on the internal email thread.

andrum99, 2020-06-04:
I'm wondering if it is perhaps worth reproducing Gordon's blog post in the documentation. Blog posts sort of get lost after a while.

aallan, 2021-07-11:
This might get done as part of #1911, but in any case, closing.

closed 2021-07-11 by aallan