raspberrypi/documentation #1548: Possibility for extension

Chewbury, 2020-05-17:
The usage of "client1" throughout this suggests the possibility of also running clients2,3,4...X this way. While doing this kind of thing (routinely) in Solaris, clients would be recognised (and authenticated) by MAC address. It would be useful to extend this documentation to add the configuration of a second (unique) client.

I intend implementing this over the next few weeks, so I'm happy to extend the page if anyone would thing it worthwhile. Yes, I'm going to run the world's least powerful cluster :)

lurch, 2020-05-18:
Sorry, which page in the documentation are you referring to??

aallan, 2020-08-24:
Closed due to lack of response.

closed 2020-08-24 by aallan