raspberrypi/documentation #1552: Pi HQ camera documentation incorrect

Adam-Langley, 2020-05-19:
I've just received my HQ camera with the intention of following these instructions to remove the filter.

https://github.com/raspberrypi/documentation/blob/master/hardware/camera/hqcam_filter_removal.md

However, removing the hex screws has not effect on the fixing the housing to the PCB. It appears the manufacturer has begun to glue the housing to the PCB. Please confirm?

Thanks

JamesH65, 2020-05-19:
@simon-martin-rpi @davidplowman Any thoughts?

davidplowman, 2020-05-19:
I'm afraid I don't know anything about that... sorry!

simon-martin-rpi, 2020-05-19:
I don't have an account so can't reply there. The gasket is rubber and
causes the mechanical assembly to act like a suction cup on the PCB. It is
not glued. Just needs a bit of force.

Simon

simon-martin-rpi, 2020-05-19:
Sorry I do have an account :-)

JamesH65, 2020-05-19:
@Adam-Langley Has that sorted out the problem? Please close the issue if so.

Adam-Langley, 2020-05-19:
Hi @simon-martin-rpi, @JamesH65 

Thanks guys - your affirmation gave me the confidence to apply more force. I was really concerned about damaging the PCB surface. (I work in software - do a lot of issue diagnostics - so I have a very well developed "make no assumptions, confirm all facts" mentality :)

I see the gasket is quite significant, and a little bit sticky - which caused it to bond quite firmly.

Additionally, I got the filter out without damaging it - a little bit of isopropyl, and warming it a little (30-40 degrees), then pushing at the very edge worked. I might fork and describe my process for anyone else concerned...

Thanks for the help guys, greatly appreciated!

closed 2020-05-19 by Adam-Langley