raspberrypi/documentation #1557: The parameter 'ip' may be helpful for RPi used out of the box

ingank, 2020-05-21:
Some inconsistent information about the use of the kernel parameter 'ip = foo' is circulating in WWW. It would be nice to have secure knowledge at https://www.raspberrypi.org/documentation/configuration/cmdline-txt.md

My favorite link is (sorry, the best i found are in german language):
http://www.netzmafia.de/skripten/hardware/RasPi/RasPi_Install.html#initip

This document explicitly deals with the various fields of the parameter as they are:

`ip=<client-ip>:<server-ip>:<gw-ip>:<netmask>:<hostname>:<device>:<autoconf>`
* client-ip
  * the IP the Raspberry should get
* server-ip
  * the IP of a netboot server
* gw-ip
  * the gateway IP, usually the IP of the router
* netmask
  * netmask of client ip
* hostname
  * host name of the Raspberry Pi, via which it would be accessible in the LAN.
* device
  * the network device for which this assignment should apply
  * for the cable always eth0
* autoconf
  * refers maybe to [RFC3927](https://tools.ietf.org/html/rfc3927)
  * allows the automatic configuration to be activated / deactivated

Example:  
`ip=10.10.1.200::10.10.1.1:255.255.255.0::eth0:off`

It would be nice if others were convinced of the importance of this parameter :)

burtyb, 2020-05-21:
FYI https://www.kernel.org/doc/html/latest/admin-guide/nfs/nfsroot.html would probably be a more authoritative link for "ip=" setup.

lurch, 2020-05-21:
@burtyb Haha, I was about to post that same link too :wink: 

@JamesH65 any idea if the `ip` kernel cmdline parameter is widely-used by Raspberry Pi users? Probably no worth rehashing the official kernel.org docs?

JamesH65, 2020-05-21:
I have no idea of the amount of use on Pi, not much I suspect, I do not remember it being mentioned anywhere.

ingank, 2020-05-21:
@JamesH65 now there is someone, remember it in the distant future! ;-)
and thx @burtyb - this is a really authoritative source :-" 
thanks to all and have a nice day!

closed 2020-05-21 by ingank