raspberrypi/documentation #1569: [Pi 4B] Datasheet register address overlap between gpio and aux

hvenev, 2020-05-27:
From "BCM2711 ARM Peripherals", Version 1, 5th February 2020

Section 2.1 claims that the so-called auxiliary registers start at 0x7E21 5000:

> Also all three are controlled by the Auxiliary enable register. The Auxiliary register base address is 0x7E21 5000.

Section 5.2 claims that the bcm2711 GPIO is mapped at the same address:

> The GPIO has the following registers. All accesses are assumed to be 32-bit. The GPIO register base address is 0x7E21 5000.

I think the GPIO is at 0x7E20 0000. That's what the Foundation linux kernel device tree says, at least.

lurch, 2020-05-28:
You're right; there's a mistake in that document and it should have read
> The GPIO register base address is `0x7E20 0000`.

Well spotted! :eyes: 

@aallan I'll send you an email about this tomorrow.

aallan, 2020-05-28:
> @aallan I'll send you an email about this tomorrow.

👍

lurch, 2020-12-12:
This has now been fixed in the updated datasheet at https://datasheets.raspberrypi.org/bcm2711/bcm2711-peripherals.pdf

closed 2020-12-12 by lurch