raspberrypi/documentation #1570: [Pi 4B] Schematics for GPIO pins 28-57

hvenev, 2020-05-27:
So far I've guessed the following, mostly based on `/soc/gpio@7e200000/pinctrl-names`:

- GPIO28-29: the ethernet PHY (using RGMII)
- GPIO30-33: bluetooth (using UART)
- GPIO34-39: wifi (using SDIO), except that by flipping "bit 1" (the least significant bit?) in an undocumented register at 0x72e0 0000 they can talk to the SD card somehow (SDIO again?)
- GPIO40-41: headphone jack (PWM)
- GPIO42: the green LED, but the VPU might be messing with it?
- GPIO43: no idea, something about some sort chip select on some sort of SPI flash?
- GPIO44-45: the internal boot ROM?
- GPIO46-57: the ethernet PHY (RGMII), but the pins appear to be configured as input pulled down rather than alt*?

How correct are my guesses?

cleverca22, 2020-05-30:
https://docs.google.com/spreadsheets/d/1m27caMlk2gofswU9ZlBD_y3y81Y_0ARSmmQgm0i1AdQ/edit#gid=0
ive been documenting some of that in here

and some other docs i have on the OTP registers
```
  reg17:
    bit 4-6, spi boot pins
      0 = none
      1 = gpio 7-11 alt0
      2 = gpio 35-39 alt0
      3 = gpio 40-43 alt4 (pre-burned)
      4 = gpio 46-49 alt2
```
so it looks like the SPI clock indeed is on the same pin as the LED!

that now makes me wonder, if theres any chance the status_led activity from linux, might be fighting the flashrom mode of `rpi-update-eeprom`, and breaking the clock?

hvenev, 2020-05-30:
> that now makes me wonder, if theres any chance the status_led activity from linux, might be fighting the flashrom mode of `rpi-update-eeprom`, and breaking the clock?

The LED code writes to the GPSET and GPCLR registers, right? If SPI is enabled, the pins are configured to ALT4, so shouldn't the LED driver's writes be ignored?

cleverca22, 2020-05-31:
ive also discovered something else interesting, the maskrom is setting 40-43 into alt4, to load the code in the eeprom
but the eeprom code is then setting 40-45 into alt4, allowing 2 more chipselect lines to do something

hvenev, 2020-06-02:
It appears that Pi 4B revisions < 4 have another EEPROM for the VL805 firmware, as explained in https://github.com/raspberrypi/rpi-eeprom/commit/fd2ea72b2677504f41298c9137647aa057f67f47

devbharat, 2020-06-02:
> GPIO34-39: wifi (using SDIO), except that by setting bit 1 << 1 in an undocumented register at 0x72e0 00d0 they can talk to the SD card somehow (SDIO again?)? Does wifi have a mux as well?

Where did you find the 0x72e0 00d0 register? How would you write to it?

hvenev, 2020-06-02:
I only heard some rumors at https://github.com/raspberrypi/documentation/issues/1209

JamesH65, 2020-06-02:
> It appears that Pi 4B revisions < 4 have another EEPROM for the VL805 firmware, as explained in [raspberrypi/rpi-eeprom@fd2ea72](https://github.com/raspberrypi/rpi-eeprom/commit/fd2ea72b2677504f41298c9137647aa057f67f47)
> 
> Is it possible that this EEPROM is also connected via SPI with one of the other chip select lines?

Not quite. Latest Pi4 board do not have the VLI EEPROM, so instead of the VLI chip loading its firmware from its own EEPROM, AIUI, it's loaded by the bootloader from a copy of the VLI firmware in the bootloader EEPROM.

devbharat, 2020-06-02:
If I wanted to use sdio on compute module on pin 34-39, for either wifi or sdcard, would I need to do something in addition to 

dtoverlay=sdio,poll_once=off,gpios_34_39

like write something to 0x72e0 00d0 ?

hvenev, 2020-06-06:
> If I wanted to use sdio on compute module on pin 34-39

There is a Pi 4-based compute module? I can't find anything about it.

JamesH65, 2020-06-06:
> > If I wanted to use sdio on compute module on pin 34-39
> 
> There is a Pi 4-based compute module? I can't find anything about it.

Not as yet.

cleverca22, 2020-07-13:
@hvenev the VL805 eeprom(when present) is connected directly to the vl805 itself

the only way to read/flash it, is to use the SPI controller inside the vl805 over the pci bus

there are currently 3 ways to flash it:
* the official `recovery.bin`
* the official `vl805` binary (static linux-arm binary) from the rpi-eeprom repo
* some flashrom patches that are in the works

JamesH65, 2020-11-19:
Is there anything being asked for here that is not in any of the datasheets? I'm a bit unclear as to what is being requested, and where it would go.

hvenev, 2020-11-21:
Documenting the register at 0x72e0 00d0 would be nice. I also couldn't find the microSD card slot in the schematics.

lurch, 2020-11-21:
> Documenting the register at 0x72e0 00d0 would be nice.

In #1209 Phil's already said "We've published all we can at this time. This thread contains the best information available."

>  I also couldn't find the microSD card slot in the schematics.

If it's really so important for you to know this, I think you should be able to work it out from the `.dts` and `.dtsi` files in https://github.com/raspberrypi/linux/tree/rpi-5.4.y/arch/arm/boot/dts and / or by looking at the CM4 IO board schematics? https://www.raspberrypi.org/documentation/hardware/computemodule/schematics.md

cleverca22, 2020-11-22:
from what ive been able to gather, only the lower 2 bits in `0x72e0 00d0` have any effect
bit 0 controls if the uSD slot goes to EMMC2(the ddr capable sdhci) or EMMC (the old sdhci), this bypasses the normal gpio altfunctions, and the uSD isnt on gpio pins

bit 1 controls if the RGMII hijacks a bank of gpio pins, bypassing the normal altfuncs, but every 2711 has a phy, so there is nothing useful you can do via changing it
it would have been nice if the CM4 exposed those pins for an external phy or more gpio, but its too late to change things now

aallan, 2021-06-07:
As per #1209 I'm closing this one.

closed 2021-06-07 by aallan