raspberrypi/documentation #1575: Update schematic for HQ camera module with dimensions for threaded holes

ajacocks, 2020-05-28:
The page hardware/camera/mechanical/rpi_MECH_HQcamera_1p0.pdf is handy, for those constructing attachments for the module. However, it is missing the dimensions for the threaded holes. This is particularly important for the back focus lock screw, which appears to be an M2 thread.

JamesH65, 2020-05-28:
@simon-martin-rpi Hi Simon,  I know you are very busy, so no hurry on this one!
@ajacocks Thanks for the report, but the guy who would do this is snowed under at the moment so may take a while if we decide it needs doing.

JamesH65, 2020-11-19:
@simon-martin-rpi Any news on this one?

aallan, 2021-06-30:
This drawing now lives at https://datasheets.raspberrypi.org/hq-camera/hq-camera-mechanical-drawing.pdf. Moved due to #1911.

akohlsmith, 2021-07-22:
Good evening - the drawing still does not have the location and dimensions of the two threaded holes that attach the mount to the PCB that holds the image sensor and electronics. I'm not sure if this was part of the initial ask, but really should be part of this drawing.

aallan, 2021-07-22:
Nope, sorry. That hasn't made it to the top of the things to do list yet.