raspberrypi/documentation #1588: Incorrect RTIMULib.ini location

woodno, 2020-06-07:
Using a Raspberry Pi 3B the following instructions on 
this page did not work...
hardware/sense-hat/README.md
<pre><code>rm ~/.config/sense_hat/RTIMULib.ini
sudo cp RTIMULib.ini /etc
</code></pre>
Removing rm ~/.config/sense_hat/RTIMULib.ini caused any program  related to calling
<pre><code>
sense = SenseHat()
bearing = sense.get_compass() 
</code></pre>
to just hang.
All the other instructions worked fine but I also needed to add this instruction to get it to work...
<pre><code>
sudo cp RTIMULib.ini ~/.config/sense_hat
</code></pre>

JamesH65, 2020-06-07:
@XECDesign Have some locations changed for the sense HAT libs?

XECDesign, 2020-06-11:
Nothing has changed, but that section of the page was copied from an old forum post, so I can't say with certainty that it's all correct.

If I recall correctly, the calibration program will give you a new `RTIMULib.ini` in the current working directory. The sense hat library will use ~/.config/sense_hat/RTIMULib.ini if it's available, or fall back to a global /etc/RTIMULib.ini.

So, I have no idea why it would hang when used from /etc or why you'd use `sudo` to copy a file within your home directory.

Not sure when I'll get a chance to look at it, but I should have a Sense HAT here somewhere.

JamesH65, 2020-11-19:
@XECDesign Did you have a chance to look at this? I might has a SH somewhere, so might be able to check it here if you have not been able to.

XECDesign, 2020-11-24:
Just gave it a go. Couldn't reproduce the issue.

JamesH65, 2020-11-25:
@woodno Can you update us on whether you are still seeing this issue?

XECDesign, 2020-11-25:
Step-by-step instructions on how to get into that state from a clean image would help a lot. I ran through the calibration steps, moved the file as documented, then ran your compass read snippet and it worked as expected.

woodno, 2020-11-25:
Hi James,
I would need to start afresh to get the issue again.
I have had the issue twice.  Each time I have started with a new Noobs.
If you cant recreate the issue then please close it.
I thought I was being helpful.  If it is only me with the issue then please
close.

aallan, 2021-08-13:
Can't recreate issue.

closed 2021-08-13 by aallan