raspberrypi/documentation #1591: Overzealous email protection

gszy, 2020-06-09:
[device-tree.md](https://github.com/raspberrypi/documentation/blob/430b92f4f31b242a83e843adba069b142904fe68/configuration/device-tree.md) contains the following string: `/soc/i2s@7e203000/status`. Some part of the website engine wrongly recognizes `i2s@7e203000` as an email address and [converts it to the following HTML](https://www.raspberrypi.org/documentation/configuration/device-tree.md):

```html
<a href="/cdn-cgi/l/email-protection" class="__cf_email__" data-cfemail="e38ad190a3d486d1d3d0d3d3d3">[email&#160;protected]</a>
```

At the very bottom of the rendered page, a script is inserted (`email-decode.min.js`) that should make the above human-readable, iff the user has JavaScript enabled.

JamesH65, 2020-06-11:
@grega FYI

grega, 2020-06-11:
Thanks for the heads up, deploying a fix...

grega, 2020-06-11:
Deployed, and can confirm that the strings are being rendered as expected both with JS enabled and disabled.

closed 2020-06-11 by gszy