raspberrypi/documentation #1596: is display_lcd_rotate still a valid property?

zjgoodman, 2020-06-20:
Hello, I have an LCD touchscreen connected to the board of my pi. The display is rendering upside-down for me so I edited `/boot/config.txt` and per [the documentation](https://www.raspberrypi.org/documentation/configuration/config-txt/video.md) set `lcd_rotate=2` and was able to fix it. However I also noticed that there's a section in the documentation that describes the property `display_lcd_rotate`. Based on the description provided in the documentation, this property seems to do the same thing as the `lcd_rotate` property. However, when I set `display_lcd_rotate=2` in my config.txt nothing happened with the display. Is this an old/deleted property that should be removed from the documentation? It's unclear if it's only supported in a particular version.

Here's my firmware version
```
pi@Tohru:~ $ uname -a
Linux Tohru 4.19.118-v7l+ #1311 SMP Mon Apr 27 14:26:42 BST 2020 armv7l GNU/Linux
```

lurch, 2020-06-21:
Are you using the Pi4? See https://www.raspberrypi.org/documentation/configuration/display_rotation.md

zjgoodman, 2020-06-22:
Ok so per [the documentation that you linked](https://www.raspberrypi.org/documentation/configuration/display_rotation.md), looks like `display_lcd_rotate` is a legacy driver property:
> Legacy graphics driver (default on models prior to the Pi4).

And [the config.txt video options documentation that I originally linked](https://www.raspberrypi.org/documentation/configuration/config-txt/video.md) doesn't mention anything about that:
> Use `display_lcd_rotate` to rotate or flip the LCD orientation. Parameters are the same as `display_hdmi_rotate`.

And the documentation for `display_hdmi_rotate` does have this paragraph:
> If using the VC4 FKMS V3D driver (this is the default on the Raspberry Pi 4), then 90 and 270 degree rotations are not supported. The Screen Configuration utility provides display rotations for this driver. See [this page](https://www.raspberrypi.org/documentation/configuration/display_rotation.md) for more information.

In my opinion the information about `display_lcd_rotate` being a legacy driver property is _discoverable_ via the documentation but not obvious. If I could propose a change, I'd suggest adding _Legacy graphics driver (default on models prior to the Pi4)_ to the `display_lcd_rotate` property's [video config.txt documentation](https://www.raspberrypi.org/documentation/configuration/config-txt/video.md) 👌

zjgoodman, 2020-07-01:
@lurch I opened PR #1606 proposing some minor rephrasing of that property's description to indicate that it's a legacy driver property

zjgoodman, 2020-07-06:
Resolved in #1606

closed 2020-07-06 by zjgoodman