raspberrypi/documentation #1614: [Pi 4B] enable_uart / overclocking documentation unclear

bcoppens, 2020-07-07:
Based on `uart.md`, one gets the impression that setting `enable_uart=1` fixes the core clock to 250MHz. However, it appears that on the Pi 4B, enabling this option appears to fix the core clock to 500MHz.

Furthermore, `overclocking.md` explicitly mentions that changing/setting `core_freq` on the Pi 4 is not allowed. If `enable_uart=1` is indeed supported on the Pi 4 (with a 500MHz clock), maybe `enable_uart` could also be mentioned in `overclocking.md`

andrum99, 2020-07-09:
> Based on `uart.md`, one gets the impression that setting `enable_uart=1` fixes the core clock to 250MHz. However, it appears that on the Pi 4B, enabling this option appears to fix the core clock to 500MHz.

I've confirmed this on my Pi 4B. It looks like the documentation needs updated for Pi 4. No idea about the overclocking - I don't overclock.

JamesH65, 2020-11-18:
@pelwell @popcornmix What's the current situation with uart and this configuration item?

pelwell, 2020-11-18:
If the mini-UART is enabled in Device Tree then the core frequency is fixed at the maximum for the platform.

aallan, 2021-06-07:
What's the status of this?