raspberrypi/documentation #1623: Bootmodes: network booting, USB booting maybe others need a refresh

mwrich4, 2020-07-15:
This section states:
"Network booting is available on Raspberry Pi 3B, 3B+, and 2B v1.2 only."

The forums claim that this is finished and no longer in beta.

The status of USB booting seems to have been improved too but the forum also unclear if any current methods are stable.  Is there also a stable method for this now as well?  

Is there a known single preferred recipe for either USB or network booting?

JamesH65, 2020-07-15:
Does this do much of what you are looking for ?

https://github.com/raspberrypi/documentation/pull/1619

mwrich4, 2020-07-15:
It looks more useful. Thank you.

In my experience, the first line of developing with a new platform has always been to scour the original manufacturer documentation for definitive statements that may derail my assumptions.  Salespeople and locker-room chat aren't always reliable since they are informal.

I believe the knowledge in a corporation should outlive any employees or forum threads.  So, keeping the documentation up to date should be a notable consideration.

JamesH65, 2020-07-15:
If you look at the Github history you will see that changes are made to the docs almost every day - so yes, we do take it seriously, and they should be fairly up to date. However there are manpower limitations to what we can do given workload elsewhere, and we don't have time to scour our own docs for things that may have gone out of date, so reports such as yours are useful.

With regard to this particular change, it was written by the only engineer who works on that particular part of the code, so I would hope it is complete - we don't have any fresher eyes than the person who wrote the code.

lurch, 2020-07-16:
> The status of USB booting seems to have been improved too but the forum also unclear if any current methods are stable. Is there also a stable method for this now as well?

https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2711_bootloader_config.md says:
"N.B. There is now a stable release of the bootloader which supports USB MSD and will only be updated with major bug fixes. Since the GPU firmware must also be upgraded USB-MSD boot as a feature is still viewed as beta software."

mwrich4, 2020-07-16:
timg236 [forum post - July 9, 2020](https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=250192&p=1527468&hilit=netboot#p1693317)
"Network boot was out of beta back in September - USB boot will be out of beta in the next few weeks. Obviously, there will be more bugs to fix and subsequent releases still go through beta->stable->production (+ critical updates)."

Do you see my confusion ?

lurch, 2020-07-17:
I realise this is slightly confusing, but there's a difference between "USB boot support in (just) the bootloader" (which went stable on [15th June](https://github.com/raspberrypi/rpi-eeprom/releases) ) and "USB boot as a feature" (i.e. updated non-beta firmware also included in Raspberry Pi OS images).

As always, patience is a virtue :wink:

mwrich4, 2020-07-17:
Do I need to be the one to close and comment?   I'm satisfied that a refresh has occurred.

I thought a more experienced person might review things and take the honor.

closed 2020-07-17 by mwrich4