raspberrypi/documentation #1633: Documentation for default terminal

nemasket, 2020-07-27:
LXterminal is the default terminal on the menu.  It is explicitly listed as the command executed.

While trying to get RVM to behave (run terminal as a login), I wound up doing the following.
NOTE:  If there is only one possible choice, no menu will be offered; the terminal software needs to be installed, e.g. 

**sudo update-alternatives --config x-terminal-emulator**

Select the number of a term (I previously installed gnome-terminal because it is known to work with RVM).
From a terminal run:

**x-terminal-emulator**

This opens a new terminal of the type you have indicated on the menu.
For RVM users, select the terminal preferences, and enable the 'run as a login shell' option.
Ok, that solved my immediate problem.  This info can be useful to others who are configuring their system to use RVM.

More importantly, to make this stick, I selected the Terminal option in the menu, right-clicked to edit and changed the hard-coded lxterminal to x-terminal-emulator.  Note that by default this was set to lxterminal, so no behavior is changed unless it is done through the update-alternatives interface.  

Now the menu item will always follow the system settings.

Documenting how to change the terminal from the default wasn't something I came across during the research of my RVM issue.

The hard-coded lxterminal vs. using the alternatives interface in the menu item is probably an issue for elsewhere.

JamesH65, 2020-07-27:
@spl237 Is there anything to do here to change the terminal on the menu if update-alternatives is used? I'm not in the slightest bit familiar with any of this.

spl237, 2020-07-27:
We recommend lxterminal - it's what is covered in all our documentation. If someone installs a different terminal program, it should appear in the menu under its own entry. So I don't think it's necessary to document this change - the vast majority of users will stick with lxterminal.

nemasket, 2020-07-27:
I can appreciate keeping it the default.  But the x-terminal-emulator uses indirection to the selection.  Which is LXterminal by default.
 
Terminal != LXTerminal.  With the new capabilities of the 8G model, there may be an increasing number of folks that find the inability to configure it as a problem.  For those people that do, it's a bit of a sideways slog to discover that Terminal and LXterminal are considered synonymous.

spl237, 2020-07-27:
For the purposes of Raspberry Pi OS, Terminal does indeed = LXTerminal and will continue to do so. It's arguably more confusing for many users to install what they believe to be an independent new program and find that one of their existing menu entries is now pointing to something different.

nemasket, 2020-07-28:
No argument,,, it's just that you've hard-coded your opinion and belief
that it is and should be lxterminal.  It is reasonable to expect the
mechanism that changes the default to work.  My comment is that if the menu
default entry is x-terminal-emulator you will continue to get lxterminal as
always.  No difference.  If you happen to change it, it would naturally
follow.  Both types of users are satisfied.

At the end of the day, I was just sharing this with you so a certain
segment of the user community doesn't have to track down the issue if they
happen to use the Pi in a certain way.  I'm satisfied since I worked it out
:)

Thanks

aallan, 2021-06-07:
> For the purposes of Raspberry Pi OS, Terminal does indeed = LXTerminal and will continue to do so. 

Closing.

closed 2021-06-07 by aallan