raspberrypi/documentation #1634: Audio setup docs do not cover well USB speakers

ivelin, 2020-07-28:
The [audio setup document](https://www.raspberrypi.org/documentation/configuration/audio-config.md) covers the headphone jack AUX and HDMI outputs, but doesn't say anything about USB speakers. Only mentions use of RPI GUI desktop to pick USB device. My use case is for a headless IoT device with Raspberry OS Lite. No desktop GUI.

I've spent hours trying to figure out why USB speakers from Adafruit with thousands of 4+ star reviews are not working consistently on RPI4 with latest Raspberry OS. See detailed problem report on the Adafruit forum:
https://forums.adafruit.com/viewtopic.php?f=50&t=167759#p821409

Any pointers that can help get to the root cause would be appreciated.

Thank you!

Ivelin

JamesH65, 2020-07-28:
Our documentation is really just for our own hardware, not for describing how other peoples hardware can be made to work. I'd suggest for technical support on this that you use the forums.

ivelin, 2020-07-28:
@JamesH65 Thank you for the prompt response.

Let me rephrase my question. I am looking for documentation that covers general use of USB speakers with Raspberry Pi and Raspberry OS. Currently the documentation talks about settings for HDMI and audiojack. It briefly mentions USB, but doesn't really specify the recommended way for enabling USB audio via command line.

Historically USB audio configuration files and tools for RPI have been shifting from one OS version to the next. There are threads about it on the RPI and other public forums:
https://www.raspberrypi-spy.co.uk/2019/06/using-a-usb-audio-device-with-the-raspberry-pi/
https://learn.adafruit.com/usb-audio-cards-with-a-raspberry-pi/updating-alsa-config

However neither seems to work for the latest OS, which is why I am looking for the official recommended way of configuring USB audio with RPI OS Lite.

Hope this makes sense.

JamesH65, 2020-07-28:
I don't believe we do anything different than any other linux, so I would expect instructions for Pi to be the same as any other similarly versioned distros.

JamesH65, 2020-07-28:
i.e. We don't have any specific instructions.

lurch, 2020-07-29:
Is there anything that @spl237 wants to add to the conversation?

lurch, 2020-07-29:
Although if using the GUI tools works, perhaps the best option will be to work out what config the GUI tools are modifying, and then try replicating that config on Raspberry Pi OS Lite? :shrug:

spl237, 2020-07-29:
Nothing to add - I don't use the CLI. Nothing that affects this should have changed in recent versions - USB devices will appear as ALSA devices (assuming they are recognised) and the relevant changes should be made to ~/.asoundrc to set as the default, or the relevant ALSA device number should be added to the command lines of applications which use audio out. As James points out - we don't do anything different for ALSA than any other Linux.

ivelin, 2020-07-29:
Thank you for taking the time to respond. I still haven't determined the root cause on either of the two systems. Both continue to have inconsistent behavior between reboots.

If and when I find out what the problem is, I will share with the RPI community.

JamesH65, 2020-11-19:
Closing as a Won't Fix.

closed 2020-11-19 by JamesH65