raspberrypi/documentation #1641: Automation script to netboot from RPi4b

OkenKhuman, 2020-08-04:
I tried to netboot RPi3b from RPi4b(as server). I followed the [official tutorial](https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/net_tutorial.md) and made some modifications like ```sudo rsync -xa --progress --exclude /nfs /media/pi/rootfs/ /nfs/client1``` and ```cp -r /media/pi/boot/* /tftpboot``` to copy the RPi3b's SD card root and boot content. And go for the remaining . But I didn't get the RPi3b bbooting yet.

So please add an automation script or GUI interface to enable network boot RPi(s) from RPi4b(as server).

lurch, 2020-08-04:
Is https://www.raspberrypi.org/blog/piserver/ the sort of thing you're looking for?

OkenKhuman, 2020-08-05:
Yes. Thanks you for the reference.

closed 2020-08-05 by OkenKhuman

lurch, 2020-08-05:
@JamesH65 @timg236 Perhaps piserver should get a more prominent mention in the docs?

JamesH65, 2020-08-05:
I know nothing about it, so not sure I could add anything useful.

lurch, 2020-08-05:
LOL, fair enough :slightly_smiling_face:  (I also know nothing about it)