raspberrypi/documentation #1642: Documentation needed to revert back rpi3b to original booting state

OkenKhuman, 2020-08-04:
I made a try to network boot RPi3b with RPi4b as server but fail. After that when I try to boot my RPi3b from sd card it is getting delayed by 45-75 seconds. 

How can I remoove the delay. Please describe how to revert back.

lurch, 2020-08-04:
What's on the SD card? AFAIK if the SD card is bootable (e.g. if you write the latest version of [Raspberry Pi OS](https://www.raspberrypi.org/downloads/raspberry-pi-os/) to it), then it should boot directly from that without attempting to network boot?
https://github.com/raspberrypi/documentation/blob/master/hardware/raspberrypi/bootmodes/bootflow.md

OkenKhuman, 2020-08-05:
It happens that the delay was on starting VNC after boot. Regarding SSH there is no delay after boot. 
Sorry for my ignorance.

closed 2020-08-05 by OkenKhuman