raspberrypi/documentation #1643: Compute module documentation, especially for camera setup is terrible.

Chick92, 2020-08-06:
Has anyone actually reviewed this documentation? 

Setting up 2 cameras with this guide is extremely difficult.

JamesH65, 2020-08-06:
What guide? Can you provide a link  to the page that is causing the problem?

Chick92, 2020-08-06:
https://www.raspberrypi.org/documentation/hardware/computemodule/cmio-camera.md

https://github.com/raspberrypi/documentation/blob/master/hardware/computemodule/cmio-camera.md

JamesH65, 2020-08-07:
OK, will have a think. Didn't know that page existed!

lurch, 2020-08-07:
> Didn't know that page existed!

#998 ? :laughing:

JamesH65, 2020-08-14:
@Chick92 I have a read through of the page, and whilst quite technical, all the information seems to be there (I cannot actually try stuff out because I do not have the equipment at home). Which particular part are you concerned with?

Chick92, 2020-08-14:
It reads like it was written as an after thought and nobody proof read it. 

It assumes an immense knowledge of the system.

This is the Raspberry Pi foundation, not Arch Linux. 

Specifically, the instructions for bringing up 2 cameras is incomplete - I bought the dev kit specifically to be able to use 2 cameras.

JamesH65, 2020-08-14:
So, which bit is missing? I can certainly try and improve some of the writing, knowing where the problem is will help with that. But remember than the CM is at the advanced end of the Raspberry Pi range, there is a certain expectation on those using it will have knowledge to be able to use that page. 

As an aside, the CM and the docs pages are supported by Raspberry Pi Trading, not the Foundation.

aallan, 2020-08-14:
The Compute Module is (mostly) targeted towards industrial customers, rather than hobbyists, and is intended to form the basis of commercial products. Documentation around the Compute Module and CMIO board will be somewhat abbreviated, and less detailed than documentation targeting other Raspberry Pi products. We would expect most hobbyists to use the Raspberry Pi Zero rather than the Compute Module if they need something in that form factor. 

Depending on your use case there are other hardware solutions to using multiple CSI cameras with a normal Raspberry Pi board, like the Ardumcam [Doubleplexer](https://www.arducam.com/product/multi-camera-adapter-doubleplexer-stereo-module-v2-raspberry-pi-zero-3-b-4/) and [Multi-Camera Module](https://www.arducam.com/product/multi-camera-v2-1-adapter-raspberry-pi/).

However having read through the page I agree with @JamesH65. Everything seems to be there, and it looks correct. The best place to ask for help if you're running into difficulty is the [forums](https://www.raspberrypi.org/forums/). 

The documentation seems to be complete, so I'm closing this issue.

closed 2020-08-14 by aallan

Chick92, 2020-08-14:
The documentation is not complete!!!!!!!!!!

Don’t tell me it’s complete just because you’ve had a casual look at it! 

There is no explanation as to what needs to be done to get 2 cameras up and running, other than a link to a dts file and some instructions on using jumpers. 

The dts file needs to be compiled to a dt-blob.bin file, which is Briefly alluded to in a different Sub section, but it doesn’t say how to do this. There is a rather poor example of how to do this in Different section of the documentation. A worked example should be provided here, as this is a key feature. There are many ambiguous aspects that need clarifying with this. Whilst it is possible to figure it out, someone could conceivably spend days on it, when a short paragraph or two would save a lot of effort. 

I’m fully aware of the target customer for the compute module, and whilst I’m certainly not at your level, I’m not exactly a Linux noob either. 

Aalan - Thankyou for suggesting other hardware alternatives after I’ve already dropped £100, because you can’t be arsed to fix your shitty documentation. I appreciate that.

Chick92, 2020-08-14:
Also, For anyone else reading - the hardware you suggested doesn’t allow for simultaneous operation of 2 cameras. There’s a software add on that will emulate this at a poor frame rate and poor resolution, however. But it’s not even close tot he capability of the compute module.

JamesH65, 2020-08-14:
I've made some changes, here is the PR. https://github.com/raspberrypi/documentation/pull/1647

The only real addition is an extra pointer to the DTS compiling information which seems to be the only obvious omission. The reason it's not on this page is that we do not like duplicating documentation and having it in just one place makes maintenance easier and mistakes less likely.