raspberrypi/documentation #1649: SPI Documentation Out of Date

TerryJColes, 2020-08-18:
Hi,

Over the past few days, I've been trying to understand the capabilities of the SPI Bus(es) on the Raspberry Pi.  Yesterday I raised an query on the Raspberry Pi Forums because I'd found discrepancies between this document and the information at [https://pinout.xyz/pinout/spi](url).  As a result of the responses I received, I raised an issue on the Pinouts GutHub page [https://github.com/Gadgetoid/Pinout.xyz/issues/372](url).

Since then, various contributors to both sites have agreed that there are discrepancies between the official documentation, what pinout.xyz says and what /boot/overlays/README implies you can do.  The most recent post at the Pinouts GitHub page points out that some of the information in the SPI spec was only valid in 2014.  In particular, the page says that one SPI Bus is available on the Raspberry Pi, whereas others can also be enabled using the overlays in /boot/overlays, depending on the Raspberry Pi Model.  (Splitting hairs, there are no SPI Buses at the pins until SPI0 has been enabled through raspi-config.)

Would it be possible to update the page to include the latest information, in particular:

- How many SPI Buses are available on each type of Pi.
What the functionality of each bus is, eg, number of CE pins, etc
How to use the overlays to enable these features (or a link to the necessary information).

I would do this myself, but I fear i still do not understand enough to get this right.

hoshank, 2020-09-01:
@TerryJColes   I agree. Even if someone could post links here for a better understanding or books that might have latest information. I started in this field a couple of weeks back.

hoshank, 2020-09-01:
This looks interesting https://elinux.org/BCM2835_datasheet_errata

JamesH65, 2020-09-01:
Which page ion our docs is out of date/incorrect?

TerryJColes, 2020-09-01:
The SPI page; [https://www.raspberrypi.org/documentation/hardware/raspberrypi/spi/README.md](url).

hoshank, 2020-09-01:
Can we confirm that the clock speed mentioned are from 2014/15 ?
It's  a little sad, the official documentation seemed outdated. Was thinking of reading a book now.

lurch, 2020-09-02:
> Can we confirm that the clock speed mentioned are from 2014/15 ?

I'm not quite sure what you're asking, but it looks like the [current kernel driver](https://github.com/raspberrypi/linux/blob/rpi-5.4.y/drivers/spi/spi-bcm2835.c#L1086) has a multiple-of-two clock divisor rather than a power-of-two clock divisor.

JamesH65, 2020-09-02:
I'll need to find someone who knows about SPI on the Pi's. I have no idea, never used it. Will update when I get some information.

JamesH65, 2020-09-02:
OK, so its does look like we ar e bit behind on that page. However, we currently don;t have manpower to fix it up right now. The elinux page is probably a good reference point for the moment until I have time to update things.

lurch, 2020-09-17:
Fixed by #1664

closed 2020-09-17 by lurch

TerryJColes, 2020-09-17:
Nice to see.

Thanks.