raspberrypi/documentation #1654: Pi versions not specified in TCP documentation

ivor52, 2020-08-26:
in the TCP documentation:
https://github.com/raspberrypi/documentation/blob/master/configuration/tcpip/README.md

- there is a mention of "previous versions", but does not detail which versions this refers to
- there is no mention of which OS version this new functionality does apply to

anyone reading this wanting to understand which Pi versions use which functionality, it is impossible to know.

lurch, 2020-08-26:
I guess if you don't have a `/etc/network/interfaces` file, then it's safe to ignore the whole `Previous versions of Raspberry Pi OS...` sentence?
Looking at the file's [history](https://github.com/raspberrypi/documentation/commits/master/configuration/tcpip/README.md) it was first added in October 2019 (so the "previous versions" it's referring to are obviously from before then).

And your original question would be much less confusing if you'd asked about "OS versions" rather than "Pi versions" :wink:   
(**all** versions or Raspberry Pi are capable of running the [latest release](https://www.raspberrypi.org/downloads/raspberry-pi-os/) of Raspberry Pi OS)

ivor52, 2020-08-26:
(sorry for the confusion, yes, i meant OS versions...)

and i'm not referring to not being able to figure it out from reading history, commits, dates and reverse-engineering to when OS versions were released.  i am of the strong opinion that documentation should be as unambiguous, obvious and stand-alone as possible.  and when referring to things such as previous releases, should explicitly state which releases it's referring to.

and that if the file isn't there isn't really helpful either since if you do an OS upgrade, the interfaces file isn't deleted in this case.

question is: will this page be updated to explicitly state which OS versions depend on the interfaces file and which use dhcpcd making the interfaces file redundant?

this is not merely academic for me: i need to support all OS versions from jessie forward where networking is automated.  given that the solution changed in some version, i need to know which so that the automated solution will get it right.

thx

lurch, 2020-08-27:
My suspicion is that nobody at Raspberry Pi has the time to look through all the different old OS versions to try and work out when exactly this sort of stuff changed. (unless @XECDesign can work out from his release notes when this all changed around?)
I suspect the only "fix" that will be put in place will be removing the whole `Previous versions of Raspberry Pi OS...` sentence if they decide that the "previous versions" are now so old that it's no longer worth mentioning the differences.

> i need to support all OS versions from jessie forward...

I'm afraid you'll be on your own with that... the documentation in this repo only covers the latest OS release downloadable from https://www.raspberrypi.org/downloads/ (which is currently Buster).

JamesH65, 2020-08-27:
The reason I used the phrase previous versions (I think it was me) was because I really didn't have time to determine on which version it may have changed. How would we specify specific versions, git hashes? If this phrase is problematic, I'm inclined simply to remove it as the docs, as @lurch has stated, only apply to the latest releases.

AIUI, if the file is there, you need to take it in to account (or delete it), if it isn't there you need do nothing.

aallan, 2020-08-27:
The point of this sentence is to let the user know that if a `/etc/network/interfaces` file is present, then it gets used rather than the `/etc/dhcpd.conf` file. The file might be present if you're working in an older version of the OS, or have upgraded from an older version of the OS rather than having done a clean install. If there is no `/etc/network/interfaces` file then there isn't an issue, if there is, you can either use that (keep your configuration) or delete it and create an appropriate `/etc/dhcpcd.conf`. 

I'm not clear on why it's relevant which version of the OS this change happened in? Or why explaining why  this file might be present, when the documentation talks about a different file, is problematic?

In theory an user who is used to using other flavours of UNIX, or has been a Linux user for many years, could easily create this file to manage their network connection without looking at the documentation and realising that the default management has flipped to `/etc/dhcpcd.conf`. I might well do this myself. Personally I hadn't realised we'd made that change.

Knowing when this change happened doesn't help your use case, as the file *might* be present in newer versions of the operating system if the user upgraded-in-place. There really isn't any way to tell except checking for the existence of the file.

ivor52, 2020-08-27:
hi guys,

let's step back a second.  for very good reasons i need to know the following information:

- which raspbian OS versions depended on the existence of the ```/etc/network/interfaces``` file?
- and from which OS version is this file no longer necessary?

yes, i can figure it out, but that's not the point.  if documentation exists which is going to describe a modification such as this, while also referring to "previous versions" where a requirement differed from the current version (or worse, some _other_ previous version), i would expect that it would be detailed in such a way as to be able to answer the above two questions.

those questions are not unreasonable, imho, and it would be very nice when documentation was detailed in the same way that programs need to be detailed in order to function properly.  why do coding standards differ from documentation standards?  (but i digress...)

cheers,

richard

p.s. and it's funny to me that for all the words that have now been put to this issue, the answers to the two questions still haven't been enunciated.

lurch, 2020-08-27:
> yes, i can figure it out, but that's not the point.

> p.s. and it's funny to me that for all the words that have now been put to this issue, the answers to the two questions still haven't been enunciated.

\*sigh\*  I'm sure if the answers to your "two questions" were as easy as you think they are, they'd have already been answered by now. The people at Raspberry Pi know as much about this issue as you; but they don't need to know this information, whereas you do... (AFAIK you're the first person to ask about this)

ivor52, 2020-08-27:
well that kind of proves my point really.

no need to sigh over frustrations regarding documentation that doesn't answer a reasonable question (in quotes? what's with the attitude?).  and i really don't understand it when you say that the ppl who are responsible for the raspbian version of the OS only know as much as i do.  they make it and don't know the details of it?  huh?  

just don't write documentation at all, then, i guess...

thanks for making never want to post here again.

lurch, 2020-08-27:
I'm sorry that you seem to be taking offence at the help that people are trying to give you.

>  i really don't understand it when you say that the ppl who are responsible for the raspbian version of the OS only know as much as i do. they make it and don't know the details of it?

As you may be aware, Raspberry Pi was released back in February 2012. In that time there have been many _many_ releases of Raspbian, by different people. Starting with Wheezy and progressing through Jessie, Stretch and now Buster (and with incremental package updates through APT as well, between major releases).
So no, I don't think there's anyone who is aware of exactly what has changed in which exact versions of Raspbian / Raspberry Pi OS. As I hinted at in my previous message, I'm sure if there _was_ somebody who knew the answers to your questions, they'd have already replied to this thread.
Ah, I've just found https://en.wikipedia.org/wiki/Raspberry_Pi_OS#Version_history which kind of proves my point.

> just don't write documentation at all, then, i guess...

This documentation is partly written by members of the community (through this GitHub repo), and partly by Raspberry Pi Trading engineers - all of whom have other (usually more urgent) stuff that they're working on as well.
In fact looking back through this repo's history, the page you're asking about was originally provided by a member of the community (who has since deleted their GitHub account, which is why their name shows up as "ghost") in #1128

JamesH65, 2020-08-28:
As one of the engineering mentioned, who is also responsible for a lot of the documentation, I do not know the answer to this question, and I do not know any one who will know the answer to the question without investigation.

I also think it might be the wrong question.  If I understand the requirements, the correct action would be to simply check to see if this specific file exists and has relevant content and action accordingly. No need for the specific OS version at all. In fact this is a much better approach than checking the OS version, because it's quite possible a user may have created this file on a more recent OS anyway, making a version check inaccurate. 

Of course, I may have misunderstood the requirements.

XECDesign, 2020-08-28:
The current and only supported release of Raspberry Pi OS (buster) has never relied on /etc/network/interfaces, although the file is there and available for people who prefer to use the legacy network setup.

I think that note would only be relevant to people running Wheezy at this point, which has absolutely no support. It's safe to remove that comment if somebody finds it confusing.

JamesH65, 2020-11-18:
OK, have removed the paragraph that is causing confusion.

closed 2020-11-18 by JamesH65