raspberrypi/documentation #1655: Clarification about "max_usb_current" deprecation

MichaIng, 2020-08-26:
The documentation has two contradicting pages about `max_usb_current`:
- https://github.com/raspberrypi/documentation/blob/master/hardware/raspberrypi/power/README.md says on RPi 2 and RPi B+ it can be used to switch between 600mA and 1200mA.
- https://github.com/raspberrypi/documentation/blob/master/configuration/config-txt/misc.md says it is deprecated and enabled by default.

The only way those fit together is when `max_usb_current=0` could still be used to lower USB current.

Does it have any effect, at least on RPi 2 and B+, either in one or the other direction?

lurch, 2020-08-26:
ping @popcornmix

popcornmix, 2020-08-27:
max_usb_current has no effect. It is enabled by default (and can't be disabled).

JamesH65, 2020-08-27:
Thanks @popcornmix, will fix the docs.

JamesH65, 2020-08-27:
Done, so closing.

closed 2020-08-27 by JamesH65

MichaIng, 2020-08-27:
Many thanks for quick answer and quick docs fix, which I just wanted to do 😄 👍.

danrneal, 2020-12-31:
I am running the latest image and kernel and this option does make a difference on my RPi model 3.  With this manually set in the config, it powers my external drive.  Without it set, it does not.

JamesH65, 2020-12-31:
Odd, it should be on by default.

popcornmix, 2020-12-31:
There is no reference to max_usb_current in the firmware code. It was removed in October 2016.
You either have very old firmware code, or you are mistaken.

danrneal, 2020-12-31:
It was a fresh image of the Raspberry Pi OS Lite (32-bit) released on 2020-12-02.  Completely fresh.  Kernel was 5.4.79.  I will reproduce again today and provide video and screenshots.

danrneal, 2021-01-01:
I'm afraid I must be mistaken because I cannot reproduce it despite running into this issue several times this week.  I imaged with Raspberry Pi OS twice (and sudo apt update right after as well) and had this issue.  Then I had the issue once again with DietPi, but now..... no issue.  My thoughts are that DietPi must have updated something that was causing my specific Pi wasn't updating with Rasperry Pi OS. :shrug:

lurch, 2021-01-01:
Perhaps your external drive is right at the limit of the power that can be provided by the Raspberry Pi, so perhaps it working / not-working is semi-random, which you happened to confuse with the `max_usb_current` setting? :man_shrugging: 
If that's the case, using a powered hub between the Pi and your external drive should make things more reliable?