raspberrypi/documentation #1659: autoboot.txt and PM_RSTS boot partition selection mechanisms are not documented

iosabi, 2020-08-30:
There's no mention in this repo about the boot process before you get to loading your config.txt.

Information that I'd like to see in the documentation that I only found buried in forum posts or running local experiments:

* autoboot.txt: Document the `boot_partition=N` option. Mention that it supports filters (this is useful even in Raspberry Pi OS to boot to a recovery partition by holding  button when things get too broken).
* Select partition to boot via PM_RSTS and how does it interact with boot_partition (my understanding is that if PM_RSTS selects the partition "0" it then looks into autoboot.txt or boots partition 1; but if you select 1 to 4 it overrides the autoboot.txt setting; but I didn't experiment with all the options). option 63 halts the CPU, but I don't know what 5 to 62 do since the bootloader doesn't seem to load extended partitions from the EBR.

I would write this as a new section explaining the boot process up to loading cmdline.txt, probably under the configuration/ directory; but if the intention is to keep this documentation very short then I'd just add an "autoboot.txt" entry to "The boot folder" page. I can write something up if needed.

lurch, 2020-08-30:
AFAIK this functionality is only useful to [NOOBS](https://github.com/raspberrypi/noobs), and isn't very useful for the general user? ping @XECDesign

It also has the added complication that on the Pi 4B the PM_RSTS only works after a "warm reboot" if you have `sdhci.debug_quirks2=4` in your `cmdline.txt`, and doing that slows down the speed of the SD card interface. See https://github.com/raspberrypi/noobs/issues/551 for more info.

But my understanding (which may well be out of date) is that PM_RSTS values of 5 to 62 **do** boot from the corresponding partition (as this is the functionality that NOOBS uses). You may find https://github.com/raspberrypi/noobs/blob/master/README.md and https://github.com/raspberrypi/noobs/wiki/Standalone-partitioning-explained and https://github.com/raspberrypi/noobs/wiki/NOOBS-partitioning-explained useful :slighly_smiling_face:

Ping also @procount in case he wants to provide information about his `rebootp` program?

procount, 2020-08-31:
I suppose this information could also be useful to anyone wriiting OS loader type utilities, or the bare metal users. I wasn't aware that autoboot.txt had filters in it. Are they same as config.txt filters?

Doesn't PM_RSTS now work on the latest Pi4s without the quirks?

PM_RSTS values do work on extended partitions, since that is what NOOBS and PINN uses,

The usual `reboot` command did not work in the NOOBS/PINN environment due to the init process, so I created `rebootp` that allows rebooting into any partition.

Additional documentation on the use of recovery.elf, start.elf, their respective priorities and the use of fixup.dat files for memory mapping would also be useful.

XECDesign, 2020-09-01:
I don't think it's useful to the average user, but it should probably still be documented. I tend to forget the exact priority order, what the limitations are and so on and having some place to refer to outside of the source code would be good. It could also be useful if somebody wanted to multi-boot without NOOBS.

iosabi, 2020-09-03:
I'm not using noobs and I find this information really useful. Even just a simple gpio filter in autoboot.txt to boot a "recovery" version can be useful (provided you do set up an extra fat partition with your recovery kernel).

Regarding the pi4 limitation I think it should be mentioned here (I filled a bug about that in the firmware repo because I didn't see the other bug in the noobs repo), however even in the rpi4 with this limitation I find this feature very useful. There are some reasonable user environment where you can still perform a warm reset

I wasn't able to make the bootloader read from an extended partition, it just wouldn't read the config.txt from there. Maybe there's a bug. Maybe there are some limitations on the extended partition support in the bootcode (like 0x5 vs 0xf extended partition format -- LBA vs CHS). Do you have a known working image with extended partitions that I can try? I don't have access to the bootcode source so all I can tell is from observed behavior =/.

lurch, 2020-09-03:
> Do you have a known working image with extended partitions that I can try?

Install NOOBS or PINN onto a large SD card, make sure your Pi has a working internet connection, and install 2 or 3 different OSes :slightly_smiling_face:

iosabi, 2020-09-04:
> Install NOOBS or PINN onto a large SD card, make sure your Pi has a working internet connection, and install 2 or 3 different OSes

Thanks for the reference! I tested NOOBS and indeed it work at least on pi3 and 4.

For the record (maybe something to hint in the documentation too), NOOBS generates an extended partition with type 0x05 in the MBR, and type 0x05 on all the second partitions of all the EBRs (except the last one of course).

Using 0x05 (Extended partition with CHS addressing) or 0x85 (Linux equivalent) as the type field of the extended partition in the MBR works fine. Using 0x0f (Extended partition with LBA) does *not* work (results in bootloader error: "Failed to open partition: NN"). However, the addressing used by the bootloader is always LBA and the CHS fields are likely completely ignored despite using 0x05. In the NOOBS-generated MBR and EBRs all the CHS start/end are set to 1023, 3, 16 except for the first partition start.

> I wasn't aware that autoboot.txt had filters in it. Are they same as config.txt filters?

I tested gpio= filters and they work. I didn't test the other ones but if the code uses the same parser I don't see why not. Maybe somebody with access to the bootcode source can confirm.

> Doesn't PM_RSTS now work on the latest Pi4s without the quirks?

PM_RSTS doesn't work if the sdcard voltage is 1.8 V. However, if you are running in an environment where the sdcard was not switched to the high speeds it works in the pi4 as well. As far as I can tell, the NOOBS OS selector screen doesn't switch the sdcard to the fast speed (which is fine for that application) so it works there. Booting a Raspberry Pi OS from NOOBS and there running "sudo reboot 6" does *not* work because Raspberry Pi OS does switch to the high speeds.

So, going back to the original bug proposal, what do you think about this and where would you like to have this documentation? I see as topics: autoboot.txt, filter to boot_partition, PM_RSTS and its precedence, recovery/kernel precedence.

lurch, 2020-09-04:
>  I tested NOOBS and indeed it work at least on pi3 and 4.

AFAIK both NOOBS and Raspberry Pi OS work on **all** models of Raspberry Pi.

Regarding the interaction of NOOBS and the bootloader - I _suspect_ that they each implement just the minimum required to make NOOBS function as expected, i.e. NOOBS uses whatever the bootloader supports, and the bootloader supports whatever NOOBS needs :wink: Which is probably why the partitioning structures used aren't as "technically correct" as they could be. (but that's just my personal guess)

Your comments about PM_RSTS and the SD card being/not-being in 1.8V mode match up with https://github.com/raspberrypi/noobs/issues/551

> does not work because Raspberry Pi OS does switch to the high speeds.

According to the comments in that issue, if you add `sdhci.debug_quirks2=4` to Raspberry Pi OS's `cmdline.txt` it would prevent the SD card from being switched into 1.8V, at which point `sudo reboot 6` _would_ work from within Raspberry Pi OS?

> where would you like to have this documentation?

Perhaps a separate page somewhere under https://github.com/raspberrypi/documentation/tree/master/hardware/raspberrypi/bootmodes makes most sense, which e.g. https://github.com/raspberrypi/documentation/blob/master/hardware/raspberrypi/bootmodes/bootflow.md and https://github.com/raspberrypi/documentation/blob/master/configuration/boot_folder.md could both link to?
ping @andrum99 as IIRC he made a lot of contributions to the bootmode pages.

aallan, 2021-06-07:
> AFAIK this functionality is only useful to NOOBS, and isn't very useful for the general user…

If it's only used by NOOBS then it's deprecated, at least in public-facing documentation. Closing.

closed 2021-06-07 by aallan

jelischer, 2022-11-21:
closing this is a bit premature..  The information is not used by other people because they can not find documentation,
leading to a circular argument..  I'm here looking for that information to write a Linux/FreeBSD multibooter. where NOOBS etc seems to assume the other OS is one of the Linux clan.. besides NOOS is not what I want. as I want the classic "switch between two already installed OS on your SD card" and I certainly don't want GUI stuff.

lurch, 2022-11-21:
There's now a **bit** of documentation about `autoboot.txt` at https://www.raspberrypi.com/documentation/computers/config_txt.html#autoboot-txt