raspberrypi/documentation #1662: Missing config option: hdmi_max_pixel_freq

acidnine, 2020-09-01:
There appears to be no official reference of `hdmi_max_pixel_freq` in the documentation although it is mentioned by Pi engineers in the forum.

Source:
https://www.raspberrypi.org/forums/viewtopic.php?t=255970#p1561253

ret394, 2020-10-15:
Please just add it to the documentation. In the HDMI mode section. 

---

1080p is just 1920×1080, though not in all cases. 'What is it for the Raspberry Pi?' Should be answered by the hdmi mode section of the page.

https://github.com/raspberrypi/documentation/blob/master/configuration/config-txt/video.md#hdmi_mode

I think the ambiguity in the table is so bad that I would remove the xp, yp values altogether. Or clarify them.

thanks

JamesH65, 2020-10-15:
I'll try and find some time to add hdmi_max_pixel_freq. We are very busy at the moment though.

Not sure what the rest of @ret394 post is referring to.

ret394, 2020-10-15:
`hdmi_max_pixel_freq` I'll add it if there's no objection. Please just focus on the more important stuff. 😊

thanks

JamesH65, 2020-11-18:
Fixed.

closed 2020-11-18 by JamesH65

ret394, 2020-11-19:
`is:open is:pr author:ret394 `

So sorry about that.