raspberrypi/documentation #1663: Adding Docker to the usage section

RyanHir, 2020-09-02:
Hello everyone,

I am curious if adding a Docker installation to the [usage](https://www.raspberrypi.org/documentation/usage/) section of the documentation. As I am sure you are aware, Docker is a containerization technology for software deployment and other tasks that may require a clean environment (a lot like a chroot). By adding this, it can introduce students into deploying their own mini lab off of their pi by using container images provided by the community. Hopefully by introducing more people to using Docker on Arm, we could see more support by maintainers to host both x86 and arm container images for possible real world deployment beyond the Raspberry Pi such as datacenters.

As I am just speculating at this point, I will finish off here and move on to what we could possibly add.

If this does become a popular idea, I am willing to work on a simple docker setup and have it run a WordPress, like we see in the [WordPress installation project](https://projects.raspberrypi.org/en/projects/lamp-web-server-with-wordpress), this will show the ease of deploying major pieces of software with prepackaged containers. And perhaps later add documentation on how to build your own Dockerfile for deployment on other people's Raspberry Pi.

Ryan Hirasaki

lurch, 2020-09-02:
I'm not sure if this is a somewhat vague topic to be adding to the Raspberry Pi documentation pages? :shrug: 

@JamesH65 Do you know if there are any Raspberry Pi engineers that use Docker on the Pi ?

The Wordpress project you link to is part of the Raspberry Pi "projects" website, and that's maintained by a different team to the Raspberry Pi "documentation" website. https://github.com/raspberrypi/documentation/blob/master/usage/wordpress/README.md  :wink:

JamesH65, 2020-09-02:
Not aware of anyone using Docker internally, and although its an interesting topic, not convinced it should be part of the docs, which are generally aimed at Pi specific stuff. i.e. topics that won't be covered elsewhere because only relevant to the Pi itself.

aallan, 2020-09-02:
It's an interesting topic, and I think it's well suited for the blog as a tutorial, or perhaps the YouTube channel. I'm unconvinced that it's a documentation issue. It's not really Pi-specific, as installing Docker on Pi is more or less the same as any other Linux system,

```
$ curl -fsSL https://get.docker.com -o get-docker.sh
$ sudo sh get-docker.sh
$ sudo usermod -aG docker pi
```

afterwards dealing with Docker and Docker images is Docker-specific. There isn't any Pi-specific documentation needed.

lurch, 2020-09-02:
Yeah, it has already appeared on the blog [a few times](https://www.raspberrypi.org/search/docker) .
ping @alexellis :wink:

closed 2020-09-12 by RyanHir