raspberrypi/documentation #1665: (Request) Kernel building documentation makes no reference to changes required for a 64bit kernel

CaptainMidnight, 2020-09-03:
This is really an observation with reference to the kernel building documentation found at: -

https://github.com/raspberrypi/documentation/blob/aallan-vpn-docs/linux/kernel/building.md

This page is specifically referencing, by default, purely 32bit kernel build procedures - it goes to some length to distinguish the different steps required to build a (32bit) kernel with regards to the various raspberry pi board versions, but makes no reference to the required changes in the various sections if a 64bit kernel is required.

As the future direction (where likely more kernel building and testing is taking place now and in the future i.e. 64bit PiOS beta) in general is surely towards a 64bit kernel, a 64bit section or more likely subsequent additional pointers to the section changes required to produce working 64bit kernels would be appreciated.

In summary, raspberry pi version specific information is documented, but kernel version (32bit / 64bit) specific variations are not.

CaptainMidnight, 2020-09-03:
... I'm fairly sure I know the required changes as I have built a working 64bit kernel, but think the document should reflect that information specifically without requiring further investigation.

JamesH65, 2020-09-03:
Something like this?

https://github.com/raspberrypi/documentation/pull/1653

We probably won't merge until we have an official 64bit release though, or at least, close to that time.

CaptainMidnight, 2020-09-03:
> Something like this?
> 
> #1653
> 
> We probably won't merge until we have an official 64bit release though, or at least, close to that time.

Hahaha yeah sort of exactly like that - OK I understand the PiOS 64bit operating system is still beta, but having a 64bit kernel isn't and enabling a 64bit kernel in PiOS 32bit is well documented i.e. it's just a config.txt variable change.

So on one hand, how to enable a 64bit kernel is fully available and accepted, yet updating the documentation which deals with rebuilding the kernel is being delayed due to the 'fully' 64bit OS (kernel and user land) being still in beta?

I do sort of understand, yet feel if you openly discuss how to enable the 64bit kernel, does that not warrant including how to rebuild it in the kernel documentation - the current situation appears to be a virtual catch22 for limited reasons.

RyanHir, 2020-09-03:
Hey, that's my PR :D.

All jokes aside, I understand where you are at @CaptainMidnight and it would still be beneficial for some users to use the 64 bit kernel on the 32 bit user space but I think it is best to wait until the OS is out of beta so further testing can be done and any possible bugs can be ironed out.

if you wish to use some form of docs for the meantime, try out [my guide from my PR](https://github.com/RyanHir/documentation/blob/master/linux/kernel/building.md).

RyanHir, 2020-09-03:
If you have any suggestions for the PR (such as config.txt changes, but not limited to) please add it as a comment there.

CaptainMidnight, 2020-09-04:
> Hey, that's my PR :D.
>
> All jokes aside, I understand where you are at @CaptainMidnight and it would still be beneficial for some users to use the 64 bit kernel on the 32 bit user space but I think [...]

Thanks @RyanHir for the link to the updated document, I did look at the link provided by @JamesH65 but as I'm still feeling my way around GitHub it wasn't obvious how to see it clearly, instead of a list of changes.

I still don't think publicly documenting how to rebuild a 64bit kernel is so out there, but I do appreciate the help confirming my steps.

RyanHir, 2020-09-04:
NP, it's kinda weird at first learning the system.

lurch, 2020-09-04:
> as I'm still feeling my way around GitHub it wasn't obvious how to see it clearly, instead of a list of changes.

You can click on the "Files changed" tab, and then on the "..." button (next to "Viewed"), and then "View file". Not obvious, but it (eventually) makes sense as you gain more familiarity with the GiHub website :slightly_smiling_face:

RyanHir, 2020-09-10:
This issue is now resolved in #1660 after being merged

RyanHir, 2020-09-17:
#1653 is merged, this issue can be closed.

lurch, 2020-09-17:
...except it doesn't yet include native 64-bit kernel building instructions, which IMHO would be useful?

CaptainMidnight, 2020-09-17:
> #1653 is merged, this issue can be closed.

Merge does not include requested material.

RyanHir, 2020-09-17:
My bad, I misinterpreted the issue

RyanHir, 2020-09-17:
Just to clarify, you want instructions to build a 64-bit kernel on device on the 64-bit OS?

JamesH65, 2020-09-17:
Reopened issue, but please be aware, we are not in "secret squirrel" mode, it's not secret in any shape or form, we simply don't have time or manpower to document it.

RyanHir, 2020-09-17:
>.... IS required by people working with the 64-bit kernel even using PiOS 32-bit BUT having the longstanding 64-bit kernel enabled via the standard config.txt configuration option.

This is currently possible with the current cross compile instructions as compiling a 64-bit kernel in 32-bit user space would technically be cross compiling and the default compiler cannot generate 64-bit binaries.

RyanHir, 2020-09-17:
but it will require some work to enable the 64-bit flag in config.txt from the instructions

CaptainMidnight, 2020-09-17:
> but it will require some work to enable the 64-bit flag in config.txt from the instructions

#1653 from what I saw was the complete instructions and solution as requested by this issue that I raised - I wasn't aware it was already being developed and available but not yet published.

It details the full steps on how to rebuild either a 32-bit or 64-bit kernel - this rebuilding could be carried out directly completely within PiOS generally speaking, a 32-bit kernel in the 32-bit PiOS and a 64-bit kernel in PiOS 64-bit although that is still within the beta programme and one of the reasons I believe the 64-bit elements have not been included/published/merged etc.

In my previous update, I was trying to highlight, even though PiOS 64-bit is still in beta that is not the only way someone may be using a 64-bit kernel ..... long before the public availability of the 64-bit beta version, the selection and use of a 64-bit kernel has been fully documented i.e. the simple change to a config.txt variable.

Therefore it's surely not hurting anyone to allow this issue 'as identified and documented' in this request to remain open as a reminder that the documentation as stands is purely aimed at the current PiOS 32-bit version with a 32-bit kernel, until deemed acceptable to allow the full #1653 text to be published.....

Please understand I'm not splitting hairs and I'm not trying to push/pressure people into publishing information that is already available but may possibly cause issues because of the still beta state of PiOS 64-bit - I personally just thought publishing the already available documented steps would potentially reduce forum questions like mine originally on the specifics required to be changed when rebuilding the 64-bit kernel.

I apologise if I'm failing to see the bigger picture or elephant in the room, just trying to help others how may want to do this from having to reinvent the wheel each time.

CaptainMidnight, 2020-09-17:
To save everybody's time and not wanting to waste resources, I've forked @RyanHir **Kernel Building documentation** that includes all the details I required.

I'm sorry if my request has been confused or confused people.

CaptainMidnight, 2020-09-18:
OK having re-reviewed all versions of the kernel building documentation, on review from a 'user' perspective, @RyanHir version of the document (branch/commit c666656d87) makes much more sense (in the real world i.e. the fact that the beta 64-bit version is widely discussed and especially now that the 'lite' build has become available) and logical flow 'if' the Raspberry Pi OS 32-bit / 64-bit labels are politically allowed to be included.

@JamesH65 Again, I do appologise if I'm over reaching, but a logical solution to this situation would be to incorporate a small change in the @RyanHir branch c666656d87 - all refernces to "Raspberry Pi OS 64-bit" become "Raspberry Pi OS 64-bit beta".

Then on full release of Raspbery Pi OS 64-bit a simple change is required i.e. 'beta' is dropped from the documentation?

IF as indicated by numerous administrator forum comments that the full release of Pi OS 64-bit is not likely to be formally released for a considerable time, maybe this is more of an acceptable approach?

Either way, technically this issue that I raised, I'm more than happy for it to be closed as the current live documenation is much more technically detailed and 64-bit related, than the version that was availble prior to raising this issue 14 days ago (i.e. August 1st commit f6d4e65 compared to the latest commit a1839d7). 👍

Thank you for making these changes to date. 👍

[Please ignore any dubious GitHub terminology in this reply]

closed 2020-09-18 by CaptainMidnight