raspberrypi/documentation #1668: Create a documentation page for EEPROM recovery

DanielBrierton, 2020-09-09:
Hi all,

Daniel here from the Digital team at RPF.

We're working on new download pages at the moment (in fact we're basically done). We no longer have the "Recovery" section that currently exists on the downloads page as we're pushing Imager more and generally making it a bit more friendly for first-time users. We do however have a section on the page dedicated to documentation where we want to add 3-4 common links. I'm looking at analytics to decide what these links should be (although that's also up for discussion), but one thing we definitely want to link to is a "My Raspberry Pi won't start up" page.

I'm happy to create a PR just taking the copy from the downloads page, but I'm not 100% sure where to put it. I figure `/setup/troubleshooting/` would be best but totally open to suggestions here.

JamesH65, 2020-09-09:
@ghollingworth FYI.

ghollingworth, 2020-09-10:
I think /setup/troubleshooting sounds like a good idea...  James, is the documentation good enough in this area for troubleshooting the Raspberry Pi?

JamesH65, 2020-09-10:
No.There no real troubleshooting at all in the docs.

 I did start a PR that was a sort of troubleshooting section, but it's
still a WIP.

It's here.

https://github.com/raspberrypi/documentation/blob/JamesH65-patch-troubleshooting/troubleshooting/readme.md

It really needs filling out then copy editing, if it's decided it's an
appropriate way of doing things.

James

ghollingworth, 2020-09-10:
I think that would be a good basis on which to start a troubleshooting system, it would be worth getting it finished, copy-edited and pushed.  It'll help a lot of people...

andrum99, 2020-09-17:
I would like to volunteer my services for this.

At the moment troubleshooting is pretty much entirely done via the forums, but it would be useful to get some basic troubleshooting in the docs, particularly as the Foundation seem to have identified this would be a useful way to proceed.

JamesH65, 2020-09-17:
Feel free to add to the Troubleshooting guide I linked above - I add to it when I have time but there are still many gaps.

andrum99, 2020-09-17:
Thanks.

It looks like what Daniel will definitely be looking for for starters is the noboot page, currently at https://github.com/raspberrypi/documentation/blob/JamesH65-patch-troubleshooting/troubleshooting/noboot.md. This is going to need some analysis of the existing troubleshooting threads on the forum to figure out a reasonably flowchart for troubleshooting the 'no boot' scenario.

JamesH65, 2020-09-18:
I did start this guide by looking through the troubleshooting thread and extracting stuff I thought was useful.

andrum99, 2020-09-18:
Indeed. It doesn't hurt to have a second pair of eyes on it though. That was actually my first reaction after I read the noboot page - it seemed incomplete.

I think we should get the Foundation's input on how to structure these pages, as it needs to be as easy as possible for noobs to understand.

JamesH65, 2020-09-18:
The docs repo is down to Trading, but we do have the comms team in Trading so they will give it a once over. But I think content first, then formatting. If we try and go via Foundation it'll be years before its get done.

andrum99, 2020-09-18:
> The docs repo is down to Trading, but we do have the comms team in Trading so they will give it a once over. But I think content first, then formatting. If we try and go via Foundation it'll be years before its get done.

Thanks - wasn't aware comms was part of trading - that's useful to know.

andrum99, 2021-03-20:
Think this issue can be closed now - the new "software" page has been live for some time now.

closed 2021-03-21 by JamesH65